christianhaitian/arkos #416: Updating the CHI GameForce test release .img

MetroidFusion, 2021-09-20:
Hi 

I have installed and running Test Release 1 for the CHI GameForce. Unfortunately, I can't seem to establish a WiFi connection through Retroarch, and the 'options -> wifi' menu doesn't work in this specific test release, as stated in the changelog. This makes it impossible for me to update ArkOS to the newest test release, since updating requires an Internet connection.

Therefore, would you consider uploading a test release .img that contains the newest version (1.3.1 as of this date) for the CHI GameForce?

Thanks so much for your hard work, and for supporting this awesome device  :)

christianhaitian, 2021-09-20:
You probably accidentally turned off your wifi with the hotkey combo.  Just press Home+Select key to toggle the wifi back on. You should then be able to see and connect to your wifi and update. https://github.com/christianhaitian/arkos/raw/main/devices/CHI_hotkeys_resize.jpg?raw=true

MetroidFusion, 2021-09-20:
Thanks for your reply and suggestion. The WiFi on my CHI GameForce should be activated, as it detects my router, and lets me enter the password in Retroarch's settings menu. But it simply doesn't connect. I have even tried connecting to different routers, with the same result, unfortunately. So I'm stuck on Test Release 1. A really good release, but I'd really like to update to the newest version:)

christianhaitian, 2021-09-20:
You sure you're entering the correct password?  If so, if you have a smartphone, try turning on the wireless hotspot feature and connect to that.  It will be a week or so before I have a new image available for this and other supported devices.

MetroidFusion, 2021-09-20:
Yes, I enter the correct password. I have actually already tried connecting it to my smartphone's wireless hotspot feature. The device detects the hotspot, and I can enter the password, but it simply won't connect/establish an internet connection. I'll hang on tight and wait for the new images, when they are published :) Again, thanks for your suggestions.

christianhaitian, 2021-09-20:
Just to be sure it's not an issue with your unit, have you tested emuelec to see if the issue happens there as well?

MetroidFusion, 2021-09-21:
It works in emuelec, as I have been able to scrape in that distro. And as mentioned, it does detect my wifi router(s), and I'm able to enter the password through Retroarch in ArkOS. But it simply doesn't connect afterwards. Weird issue, but it seems like the 'option -> wifi' menu works in your newest release for the CHI GameForce, so I'm looking forward to trying that :)

MetroidFusion, 2021-09-21:
Update: I just tried activating yet another mobile hotspot, and I finally succeeded at establishing a connection! Don't know why it wouldn't connect to my other devices, but there you go :D I have now updated the CHI GameForce, even though I'm probably gonna do a fresh install when the new .img is released. Thanks again for your suggestions and your awesome OS! :)

christianhaitian, 2021-09-21:
Strange but glad it eventually worked for you.

closed 2021-09-21 by christianhaitian

Casao, 2021-09-23:
@christianhaitian Not a CHI, but I've noticed on my 351p running final that my wifi will not connect if my security is set to WPA2/WPA3 combo, but works fine for WPA1/WPA2 combo. It sees the network just fine, just fails to ever connect to it, silently. 

It also works fine for the WPA2/WPA3 combo in 351elec, so it seems to be ArkOS specific. Happy to help debug/test that if you're interested, but this sounded very similar.

christianhaitian, 2021-09-23:
Odd as we're both using the same kernel.  WPA3 would make sense to be problematic though.

Casao, 2021-09-23:
Yeah, the WPA2/WPA3 setting is supposed to be backwards compatible with 2 completely, but I'm assuming it's something like it sends out a 3 first and falls back to 2, but the system is getting hung up negotiating the 3 and it times out. Because it does take a good long while to fail to connect.

christianhaitian, 2021-09-24:
May not matter for you now but a new image is now available.