christianhaitian/arkos #417: Request: Add .Jar Emulator such as J2ME

IFTTTexas, 2021-09-22:
I think ArkOS would be a great platform to play old mobile games on. Thanks for all you do.

christianhaitian, 2021-09-22:
Perhaps you can try this through the psp emulator and report your findings.
https://wololo.net/2016/12/15/trip-homebrew-lane-pspkvm/

christianhaitian, 2021-09-24:
Any update on this?

christianhaitian, 2021-09-28:
Good to know.  I'll see if I can have it setup as a system with those settings preconfigured.

tylibra2, 2022-01-30:
Were you able to run Gangstar on the emulator?

rhunjadi, 2022-04-27:
I have tried some java games over the pspkvm  and most of them don't work.
Would it be possible to add j2me emulator like AmberElec has it?

christianhaitian, 2022-04-27:
Last time I tried it, it wasn’t very good.  I’ll review again.

christianhaitian, 2023-10-17:
I've decided this is not something I will pursue as I have no desire for j2me at this time.  I may change my mind in the future and if I do, I will include one in a future update.

closed 2023-10-17 by christianhaitian

C4RLO5-13, 2024-02-17:
Some colleagues will be able to run Java on the Arkos, more specifically on the R36s, so I have no audio, but they are already seeing if this can be fixed.

anshulsingh8326, 2024-05-18:
Any luck? @C4RLO5-13