christianhaitian/arkos #418: redshift/blue light filter

iateadonut, 2021-09-24:
How can we get a blue light filter on this?

I tried installing 'sct' from the  but I just get "segmentation fault" if i try to run it.

I also tried installing 'redshift', but that would take 384MB, plus it looks like it needs to install xserver-xorg... so I guess ArkOS doesn't use xorg and redshift wouldn't work.

### What device are you using?
RG351P

christianhaitian, 2021-09-24:
I have no idea and not something I’m interested in further investigating.  Xorg is not preinstalled in ArkOS.

closed 2021-09-24 by christianhaitian

iateadonut, 2021-09-24:
What is used as a display server?  I'd like to see if I can figure it out myself.

christianhaitian, 2021-09-25:
KMS/DRM.  You may find more helpful information by reviewing the odroid go advance forum.  One example is here: https://forum.odroid.com/viewtopic.php?p=277231#p277231