christianhaitian/arkos #419: locale RU ArkOS 2.0

SlowPixel, 2021-09-29:
RG 351 V Russian
I have translated everything except.
Most likely, something has changed in 2.0, because everything is translated in 1.7.
#~ msgid "%i GAME AVAILABLE"
#~ msgid_plural "%i ИГР ДОСТУПНО"
#~ msgstr[0] "%i ИГРА ДОСТУПНА"
#~ msgstr[1] "%i ИГРЫ ДОСТУПНО"
#~ msgstr[2] "%i ИГР ДОСТУПНО"

File with translation by link
https://drive.google.com/drive/folders/1bD1Kr6CDYGojNAFA7NpguMzylD5r4qg6?usp=sharing

closed 2021-09-29 by SlowPixel

christianhaitian, 2021-09-29:
Thank you.  There are some additional changes coming in the next update.  See [here](https://github.com/christianhaitian/EmulationStation-fcamod/tree/351v/resources/locale/es) for the changes in case you need to add additional translations for this file.

SlowPixel, 2021-09-30:
Completed
https://drive.google.com/file/d/1VGhouVjaCyUZgvcUhbmb_G8psyEltWIn/view?usp=sharing