christianhaitian/arkos #422: RG351MP Battery Stop Charging after a couple of minutes

rymexs, 2021-10-04:
First of all, thank you very much for the effort you put into ArkOS. I've been beta testing the RG351MP and it runs excellent, but today when I wanted to charge the console, I found that it only charged for a couple of minutes and then the charging light disappeared. When checking the battery, it showed the same percentage. I tried with different cables to validate if that was the problem but the same thing happens.

christianhaitian, 2021-10-04:
In the latest update for this beta, we're attempting to use the internal power management controller to give a more accurate measure of battery.  This works well for the OGA devices and related clones.  It should hopefully work well for the Anbernic devices as well.  What has to happen so that the controller understands the correct parameters of the battery, you need to do a complete drain of the battery then recharge to full in order for the battery indicator to be correct using the internal power management controller.  

The battery measure used in the stock card as well as what was provided in source goes by voltage which is wildly incorrect as witnessed with the RG351P/M/V.  For instance, if you drain the battery on stock then plug it into a power source, it will immediately show 100%.

closed 2021-10-04 by christianhaitian

Pash-tet, 2021-10-14:
@christianhaitian 
OS: ArkOS 2.0 Test 1.7
I tested the problem with displaying the battery charge for several days, here are my observations.

1. I tried to completely discharge the battery, and then fully charge it - the charge display is stuck at 22%.
2. To fix the display functionality, I had to disconnect the battery and reconnect it. Charge display 100%.
3. After that, the charge only drops. For example, I discharged the console to 50%, then I charge it for a couple of hours, the charge will remain at 50%.

christianhaitian, 2021-10-15:
Thanks for this feedback.  I just released update 1.8 that will revert the battery measurement back to the same method used for the other rg351 devices.  It may not be accurate but it seems to be the best solution available for now.