christianhaitian/arkos #423: RG351V reports as 149 Hz unless installed from fresh 2.0 image

gnubeest, 2021-10-05:
I've heard that Anbernic sources a few different displays for the 351V, and since I haven't seen otherwise elsewhere on the internets, I'm running under an assumption that mine is slightly unique because it appears to run effectively at 149 Hz. (!)

And yes, I had a hard time believing it too, but it seems to be 640x480p149 both in /sys/class/drm and in practice. Everything I throw at it is consistently silky smooth once I turn Sync to Exact Content Framerate on, from PAL consoles to 75 Hz handhelds to arcade boards with random vertical refresh rates.

I'd like to keep it that way, but I wanted to report that while RetroArch reports and runs at that refresh rate both from a fresh ArkOS 1.7 install and one upgraded to 2.0, installing from a new 2.0 image locks it at 60 Hz for some reason.

christianhaitian, 2021-10-05:
Yes in the 2.0 image, I provided an updated .dtb in the boot partition that locks the framerate to 60hz.  Feedback I got is that it’s smoother this way.  Is this not the case for you?

gnubeest, 2021-10-05:
It's absolutely not. Even in 60 Hz games I alternate every few seconds between relatively stable vsync and crazy frame judder with the 2.0 image. (And then also lose the aforementioned clean sync on systems with non-~60 Hz refresh rates.) This display does not seem to like being forced to 60 Hz at all.

christianhaitian, 2021-10-05:
Sigh...ok.  Just copy the original .dtb from the boot partition in the 1.7 image to the fresh 2.0 and it should resolve the issue for you.  You can use 7zip to easily do this without having to flash it to a sd card.  I’ll provide an update in the near future that will allow people to switch between the new .dtb and the previous one depending on their preference.

gnubeest, 2021-10-05:
I frequently excel at being an edge case, I do apologise. I'll keep my 1.7 -> 2.0 install for now until the next time I decide to nuke it again, but I feel better knowing what's going on now in case I go tinkering. Thanks for following up!

christianhaitian, 2021-10-16:
New update is available that will allow the user to switch the screen timings to either the original (Anbernic) timings or to the updated timings.