christianhaitian/arkos #424: Name of engine used on Port not consistantly mentionned

ptitSeb, 2021-10-05:
### Describe the Issue (If applicable)
The Wiki list of the Port does not consistantly mention the Engine used to run the game. Sometimes it mentionned as main title 
 with the game inside parenthesis (like "DevolutionX (Diablo 1)"), sometimes in the comment 'like "Shovel Knight" whre box86 and gl4es are mentionned, also, typo on my name there). Sometime nothing is mentionned (like HalfLife I assume it's Xash3D, Maldita Castilla/AM2R/Freedom Planet all use box86 too I suppose).

Mentionning the engine / link the repo is not mandatory, but it helps keeping every one involved motivated.

christianhaitian, 2021-10-05:
Ok.  I will work on adding that information for all the ports mentioned there.  I've corrected your name in the wiki for Shovel Knight using box86.  Maldita Castilla and AM2R doesn't make use of box86.  I'm not sure what MC uses so I'll credit Locomalito directly but there's no box86 or gl4es being used in that one.  AM2R is using droidports used by [JohnnyonFlame](https://github.com/JohnnyonFlame/droidports).  Your work and others are appreciated so if all that's needed is a mention and a link to the repo, I see no issue with that.

ptitSeb, 2021-10-05:
Thank you!

Yes, credit Locomalito for Maldita Castilla! Also, I got many other games from him, running on the Pandora at least, like "The Curse of Issyo", Gaurodan or EFMB. I also have Maldita Castilla 1.1 running (the Windows version), using a more recent runner (you can grab the Pandora version from https://repo.openpandora.org/?page=all&search=locomalito and use 7z to extract the datas).

For all those GameMaker2 games (using runner), I was assuming it was using x86 linux version of runner with box86, but I think there is also a RPi binary of runner, so I guess this is what is used.

christianhaitian, 2021-10-05:
Thank you for this info.

ptitSeb, 2021-10-07:
Thanks for the updates in the Wiki :)

closed 2021-10-07 by ptitSeb