christianhaitian/arkos #439: RG351V WILL NOT BOOT

ultimateives, 2021-10-25:
When updating to ArKOs the BOOT screen loads the new 2.0 boot image then either goes to loading screen then while loading gets stuck. 

Ive tried 3 different SD cards, tried flashing with 2 different softwares and the issue still conts. 

When 351Elec or RA is flashed everything boots properly.

Im using a Sandisk 16G for the system and Sandisk 128G for the rom folders.

Thanks

christianhaitian, 2021-10-30:
Hi, any update on this?

ultimateives, 2021-10-30:
I thought I replied, sorry.

1.7 does boot, when updating within the OS to 2.0 no luck. Still freezes at
start.

christianhaitian, 2021-10-30:
> I wonder if it's the update from 8/27/2021 for the black lines issue. Here's how you can test this, backup the rk3326-rg351v-linux.dtb file located in the /boot location. You can do this from your computer as the boot partition is fat32 and can be modified from a Mac or Windows PC. Update to the latest 2.0 version via the OTA update. Then restore that file back to that location. See if it boots fine then.
> 
> That's the only change I can possibly see causing this issue you're having.

Did you try what I suggested above?

closed 2021-11-06 by christianhaitian

ashfrench-dev, 2022-10-18:
Hi, I feel like I'm having the issue with my RG531V. I don't know where I can find 1.7 but I'd be up to try what you suggested for OP if it helps you identify a problem.

Just trying to get 2.0 working, flashed with USB IT and I get a blinking cursor after the loading screen.

ultimateives, 2022-10-18:
I gave up on it unfortunately. Nothing ever worked and I was never able to
boot 2.0.

ashfrench-dev, 2022-10-18:
Ok. Any idea where I can find 1.7 these days?

ultimateives, 2022-10-18:
That's gonna be released by the developer only. I no longer have access to
it. Sorry.

ashfrench-dev, 2022-10-19:
Just wanted to update that I was able to get 2.0 running on the RG351V by making sure the SD card was formatted to exFAT before flashing the OS onto it.

christianhaitian, 2022-10-20:
Very odd but good to note.

nmatsuda, 2022-10-28:
I'm having what appears to be the same issue on the RG353V. I'm using USBIT and preformatted to exFAT but I can't get past the ARKOS splash screen and (sometimes) the dots. Since there isn't an RG353V build of the 1.7 version referenced above, is there anything else I can try?

christianhaitian, 2022-10-28:
What micro sd card are you using?

nmatsuda, 2022-10-28:
It's a Samsung 64GB EVO Select, so should be decent? I tried two different cards (thought they were the same make/model) and same issue on both.

christianhaitian, 2022-10-28:
You made sure you downloaded the right file for the r353v?

nmatsuda, 2022-10-28:
ArkOS_RG353V_v2.0_10022022.img?

I did just get a new error on the dots screen, it says:

`sudo: unable to resolve host rg353v: Temporary failure in name resolution` 

That appears to be a non-disk related network service issue, but either way it doesn't get past that.

christianhaitian, 2022-10-28:
jelos boots fine?

nmatsuda, 2022-10-28:
yep, as well as the stock anbernic linux, all from that same SD card

christianhaitian, 2022-10-28:
Ok.  I think I know what the issue may be.  I'll provide a file for you to try shortly.

nmatsuda, 2022-10-28:
sounds good. In the meantime I tried a different SanDisk card and had the same problem, so that confirms it's not an issue with that specific Samsung card.

christianhaitian, 2022-10-28:
Hi,

Take the attached zip file, open it and copy the rk3566-OC.dtb file in it and 
[rk3566-OC.zip](https://github.com/christianhaitian/arkos/files/9891073/rk3566-OC.zip)
to the boot partition of the ArkOS card for the rg353v.  Overwrite the existing file there.  Then see if you're able to successfully Boot ArkOS afterwards.  Let me know your results.

Thanks,

nmatsuda, 2022-10-28:
hmm no now I don't get anything at all - blank screen, no splash

christianhaitian, 2022-10-29:
Try a new flash of the sd card with that new file copied to it after the flash, then boot the sd and see if anything changes.

nmatsuda, 2022-10-29:
yeah that's what I did, fresh flash, then copy

christianhaitian, 2022-10-29:
Hmmm....I'm out of ideas at the moment then.

nmatsuda, 2022-10-29:
I'm assuming you have plenty of people successfully using this build on the rg353v? I wonder if there are some gremlins with this particular unit - I have had a bunch of issues with Android (it reboots randomly).

christianhaitian, 2022-10-29:
Yes, plenty have been able to boot ArkOS on their RG353V without issue.  Some even use it on their RG353P units which I don't provide support for.  I've seen reports lately of Android being unstable on the RG353V so not sure if it's related to this particular issue or not.  I haven't had issues with the Android side but I've only used it twice to play Duck Hunt on the NES emulator.

nmatsuda, 2022-10-29:
no worries if you don't want to keep trying to figure this out, but two new observations:
- the blank screen appears to only happen when usb power is plugged into the device (but only with the new blob file for some reason). I still get the boot logo when it's not plugged in
- I plugged a keyboard into the OTG port to see if I could Alt+F1 to another terminal. That doesn't seem to work but it will show boot messages. A bunch of the services are failing to start, specifically: networkd-dispatcher, wpa_supplicant, systemd-resolved, NetworkManager, and systemd-logind.  Now I'm not sure why a bunch of network related services would halt the boot processes entirely, but that seems to be the issue.

Is there any way I can get to a terminal (including rebuilding if necessary)?

s0len0id, 2022-11-06:
(RG351V, USBIT, original SD, with and without 2nd SD inserted)
Probably just a coincidence, but my brand new RG351V just started doing this (top-left blinking cursor hang on blank screen while booting very slowly, after splash and dots).  
I *think* i was rebooting after changing the timezone. Funny also this thread started last year around this time of (changing to wintertime). 

Reflash coming up :)

christianhaitian, 2022-11-06:
That sounds like a bad sd card

s0len0id, 2022-11-06:
i think i've narrowed it down to empty battery (the displayed % is way off, both directions). I've not had a reboot-hang yet while full or charging. Will keep reproducing.
Thanks for your quick reply, massive work, respect! :+1:

christianhaitian, 2022-11-06:
That's a known issue as well with this chipset

s0len0id, 2022-11-29:
I've narrowed down my issues to just the RESTARTING hanging with a usb-connected teensy with M8 headless firmware: https://github.com/jasonporritt/rg351_m8c/issues/9.

tallpr24, 2023-09-23:
hello all im having the same issue. brand new rg353v, cant load ark os 2.0 any fix for this?
gets pass the arkos screen, then i get a black screen.

ultimateives, 2023-10-06:
I gave up. I tried everything, new SD card and all. Developer was very
helpful unfortunately I couldn't get it to work.

rawia-larbi, 2024-03-25:
I got the same issue but here how I solve it :

1) use mini Tool partition wizard to format your SD card (if your SD card was used before) , I tried multiple different ways to format : windows tools, SD card formatter no one worked.

2) For writing you img in the SD card, use win32 disk imager and not "refus". Refus was showing only one partition : BOOT. But win32 disk imager is able to get the 3 partitions.

ultimateives, 2024-03-25:
Will have to knock the dust off and give it a try. Thanks!