christianhaitian/arkos #442: RG351MP controller issues

Kise12, 2021-10-29:
### Before submitting your issue or feature request.  DO NOT REQUEST PORTS THROUGH THIS PROCESS.  I will no longer be focusing on adding additional ports at this time.  You may try to post your request [here](https://github.com/krishenriksen/AnberPorts) for an alternative unofficial solution.  Do understand support for those ports will not be available on this github.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.
The hotkeys don't work. Such as start + select for exiting games, select + x for RetroArch menu, etc. The control scheme for Nintendo DS is off to the point where games are un playable and you can't navigate the DraStic menu to change the control scheme. For the N64 the controls don't work at all.
### How can the issue be reproduced? (If applicable)

### What device are you using?

Anbernic RG351MP

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
2.0 1.9.1.1

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2021-10-29:
You must have copied files from another system because all these work fine on my end and many others.

christianhaitian, 2021-10-29:
If you did a back and restore of settings from another 351 unit or copied your roms from another setup, don't do that as the controller on this unit is different from other 351 devices

Kise12, 2021-10-30:
Yes, I copied a backup file from my Rg351m. Thanks for the info, I wasn’t aware that that was an issue.

closed 2021-10-30 by christianhaitian