christianhaitian/arkos #443: Wifi setup missing on screen keyboard characters.

Jdehler, 2021-10-29:
The & (ampersand Alt+ 38) is missing from the Wifi setup's OSKB.
This is a nasty issue for me as the business I work for has several machines and devices that are frequently brought in but not long enough to catch the user and re-enter a new wifi key.

I wouldn't trust most of these users to make a peanut butter sandwich without the peanut butter let alone enter a new wifi key.

What would be a good workaround is if the wifi setup app could read a text file with the wifi SSID and key already there then delete the text file after asking the user for confirmation. This would only open up the wifi key to exploitation for a short window without having to add the ampersand and any other missing but valid wifi key characters to the keyboard.

christianhaitian, 2021-10-30:
You can use Retroarch to setup your wifi instead or use the alternative solution described here: https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---rg351mp#q-how-do-i-configure-my-wifi-connection-in-arkos

closed 2021-10-30 by christianhaitian

intrepidsilence, 2022-12-09:
Why not fix the keyboard?

christianhaitian, 2022-12-09:
I don't have the source code for that keyboard.  I'm hoping someone can provide a keyboard solution with source code.