christianhaitian/arkos #444: Online (OTA) Updates are currently down for except for China

christianhaitian, 2021-10-30:
gitcdn.link is currently down which is impacting ota updates at this time.  I will update and close this ticket when the ota is back up or with an update on how to resolve this.

Rudyhernandez1, 2021-10-31:
It's not working for me at all. https://user-images.githubusercontent.com/93460519/139567290-81d09026-b223-431a-b7ce-74567f870d37.MOV

christianhaitian, 2021-10-31:
> It's not working for me at all. https://user-images.githubusercontent.com/93460519/139567290-81d09026-b223-431a-b7ce-74567f870d37.MOV

You restored this file as instructed above?

christianhaitian, 2021-10-31:
If so, also make sure to enable remote services and see if that resolves the issue for you.

Rudyhernandez1, 2021-10-31:
I don't have a sd card reader for my MacBook since I got my device on Etsy with everything already loaded on.

christianhaitian, 2021-10-31:
No worries, there are many other ways to transfer the file onto your installation.  For instance, you can use arkbrowser to load the file on to the device.  See instructions in the faq for the rg351v on how to do this. https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351V#q-how-do-i-use-arkos-browser-for-managing-my-roms-via-a-web-browser

Rudyhernandez1, 2021-10-31:
Yes! I tried doing this too but it was kind of confusing for me. I'm literally brand new to all of this.

christianhaitian, 2021-10-31:
Do you understand how to use ftp?

Rudyhernandez1, 2021-10-31:
[image]

christianhaitian, 2021-10-31:
You didn't follow the instructions. :(

Rudyhernandez1, 2021-10-31:
Ahh noo :(! Yeah it's so confusing to me

christianhaitian, 2021-10-31:
do you know how to use ftp?

Rudyhernandez1, 2021-10-31:
I have no idea how to use it. What I did was clicked the link of the file then downloaded it and then created a folder since I have a 351V and pasted the download in the folder. I have my 351v plugged into the MacBook  with the charger.

christianhaitian, 2021-10-31:
I think the best solution for you will be to get a sd card reader so you can easily make the change or hope that the existing cdn comes back online.

Rudyhernandez1, 2021-10-31:
Yes! That's what I'm thinking do you recommend any good ones? And do you think it will eventually come back online? Has this happened before?

christianhaitian, 2021-10-31:
There are some cheap ones you can get either in a physical store or through Amazon.  Anyone that you can plug into your usb c port on your Macbook should be fine for this task.  This outage has happened before but not for this long length of time.  In reviewing the github for this service, there's been no activity from the owner for over 6 months so not sure if it's being maintained anymore.  The resolve I posted above switches to a new provider and adds a backup in case an outage occurs again to minimize on this issue from occurring again.

Rudyhernandez1, 2021-10-31:
Ohh okay! Yeah I'll probably just order an sd card reader do you know if I'll need to download any additional apps on my MacBook?

me038372, 2021-10-31:
The instructions worked perfectly for me. Thank you. I updated my two RGB10s in this fashion.

christianhaitian, 2021-10-31:
> Ohh okay! Yeah I'll probably just order an sd card reader do you know if I'll need to download any additional apps on my MacBook?

You shouldn’t have to.  It’s just a matter of plugging the sd card from your TF1 slot in the 351v, into the card reader, going to the easyroms folder, follow the directions.

closed 2021-11-01 by christianhaitian