christianhaitian/arkos #445: PSX vibration function is not working properly on the current RG351mp.

haveanair1, 2021-11-01:
PSX vibration function is not working properly on the current RG351mp. It only vibrates once at first and it doesn't work. I suspect the problem is not caused by removing vibrations on boot.

christianhaitian, 2021-11-01:
Just update the core via retroarch and it should fix this issue.

closed 2021-11-01 by christianhaitian

haveanair1, 2021-11-01:
The psx entry does not appear in the retro arc 32 core online update. The rumble version of the core still did not work properly, but when I changed the core to a psx core rather than .rumble, it worked normally.

haveanair1, 2021-11-01:
mgba rumble core can't get into retro arc settings.

christianhaitian, 2021-11-01:
These work fine on my end.  Did you restore settings from another unit?

christianhaitian, 2021-11-01:
You should be able to update both pcsx-rearmed cores online now.  I just needed to fix the .index-extended content in the central repo.

christianhaitian, 2021-11-01:
mgba and mgba-rumble both work fine on my end still so I suspect your retroarch configs may be the issue.  As I said previously, if you restored your settings from another 351 units such as the P or M or V, that is most likely your issue as the controls on the MP are completely different from those other units.