christianhaitian/arkos #446: Odroid go advance black edition l2/r2 buttons?

icefill, 2021-11-03:
### Before submitting your issue or feature request.  DO NOT REQUEST PORTS THROUGH THIS PROCESS.  I will no longer be focusing on adding additional ports at this time.  You may try to post your request [here](https://github.com/krishenriksen/AnberPorts) for an alternative unofficial solution.  Do understand support for those ports will not be available on this github.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
OGA 1.1 l2 and r2 buttons are not assigned properly.

A clear and concise description of what the bug is.

OGA 1.1 (odroid go black edition)'s l2 and r2 buttons not work due to they are not properly assigned. Two of function keys are assigned as l2 and r2 according to below, even though OGA 1.1 have seperate l2 and r2 buttons other then function keys. https://github.com/christianhaitian/arkos/wiki/Key-codes-and-Global-Hotkey-config-file-locations

### How can the issue be reproduced? (If applicable)

### What device are you using?

Odroid GO advance Black edition

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
2.0 (10172021)

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2021-11-03:
Interesting no one brought this up until now.  I'll have to review for a possible future fix.  I don't have a v1.1 unit, it would be helpful if you could post the results of `evtest` for l2 and r2.

icefill, 2021-11-04:
> Interesting no one brought this up until now. I'll have to review for a possible future fix. I don't have a v1.1 unit, it would be helpful if you could post the results of `evtest` for l2 and r2.
I added a key test photo. Is it the evtest?

christianhaitian, 2021-11-04:
No.  evtest looks like the following for example
[image]

icefill, 2021-11-04:
[image]
I managed to open ssh and tried evtest. @christianhaitian

christianhaitian, 2021-11-04:
Hmmm...what image did you download for use with your oga be?

icefill, 2021-11-04:
> Hmmm...what image did you download for use with your oga be?

the one named OGA 1.1/RGB10 on the arkos wiki page.

christianhaitian, 2021-11-04:
These button assignments are similar to what's used in the image.  What system(s) did you noticed issues L2 and R2 not being assigned correct?

icefill, 2021-11-04:
You mean l2/r2 buttons should work properly? I tested a psx game and shadow warrior on portmaster. I can change the default button setting on retro arch but shadow warrior had no luck.

christianhaitian, 2021-11-04:
Yes, L2 and R2 should work properly.  When you loaded the image for the first time on your oga 1.1, when it loaded emulationstation, did it ask you to setup your input?

icefill, 2021-11-04:
I tried OGA 1.0/RK2020 version first. It did ask me input setup.
Then I tried OGA 1.1/RGB10 version, It didn't ask me input setup.

christianhaitian, 2021-11-04:
The OGA 1.1/RGB10 version is the correct one you should be using and wouldn't ask you for input setup.  Then the L2 and R2 should be already correct.  Did you setup an override for your button configurations?

icefill, 2021-11-04:
> The OGA 1.1/RGB10 version is the correct one you should be using and wouldn't ask you for input setup. Then the L2 and R2 should be already correct. Did you setup an override for your button configurations?

No I did not. In shadow warrior case, there is a setting that can change controller setting. Other buttons work properly but l2 and r2 buttons are assigned to f3 and f4 while real l2 and r2 buttons didnt do anything. In psx case, I could go to retro arch setting that can change basic controller layout. In default one, function keys are set to l2 and r2 so thay I changed them to corrwspondkng button codes and they worked properly

christianhaitian, 2021-11-04:
strange but happy you got it figured out.

icefill, 2021-11-04:
> strange but happy you got it figured out.

No I mean I still can't change shadow warrior...

christianhaitian, 2021-11-04:
Then maybe something is up with that game and the psx emulator.  Have you tried duckstation just to see how the buttons work there?

icefill, 2021-11-04:
Okay I will try other emulators. I dont know anything about  firmwares and such but to me but it still looks f3 and f4 are assigned in the position of l2 and r2 to me. Thanks for making this amazing arkos and port master btw...

christianhaitian, 2021-11-04:
It sounds like it is in your case but looking at the data you provided which is similar to the rgb, it shouldn't be as the l2 and r2 button on the rgb10 are the same values

icefill, 2021-11-04:
Duckstation has the same issue. I think you can close the issue since I can workaround it by modifying retro arch settings (except Shadow warrior...) Thanks for your effort concerning the issue.

icefill, 2021-11-05:
For those who having a same problem with me, Shadow warrior buttons issue is solved by editing Shadow Warrior.sh file's sdl_controllerconfig of oga parameter.
(left_trigger:b12 to b6, right_trigger:b13 b7)

icefill, 2021-11-06:
@christianhaitian I am sorry to keep you annoying but can you explain the details of l2/r2 to me? when I saw the first time the keycodes of rgb10/oga 1.1 in the wiki are different from what I saw in my oga1.1's evtest althought it seems to be edited now. It seems that oga 1.1 and ogs have same key codes and either rgb10 have different keycode with oga1.1 or rgb10's l2/r2 buttons will not work properly either.

christianhaitian, 2021-11-06:
I fixed the documentation on the wiki as it was wrong.  The rgb10 and oga 1.1 keys for l2/r2 are the same based on what you provided along with checking ng on my rgb10 unit.

icefill, 2021-11-06:
and your rgb10's l2 and r2 buttons work properly?

christianhaitian, 2021-11-06:
Yea my l2/r2 are fine.  there a few others that use arkos with the oga 1.1 and have not reported this issue so I don't know. ¯\_(ツ)_/¯

icefill, 2021-11-06:
Okay.. ;(

christianhaitian, 2021-11-06:
Although I do see the issue with the portmaster ports for the l2/r2 buttons.  That does need correction for ports that make use of l2/r2 for the rgb10/oga 1.1.

christianhaitian, 2021-11-08:
After further research, L2 and R2 are indeed different on the RGB10 than on the OGA 1.1.  I won't change the configuration as for the OGA 1.1, it just uses a couple of the F buttons for L2/R2 that are rarely used for most games.  For the RGB10, OGA 1.1's L2/R2 don't exist at all on the unit.  Those who use ArkOS on the OGA 1.1 and want L2/R2 at the top will need to modify their setups for that or use a similar distro that specifically targets the OGA 1.1 like TheRA or RetroOZ.

closed 2021-11-08 by christianhaitian

christianhaitian, 2021-11-27:
Great news, A fellow user from the RGHandhelds forum had a great idea on how to easily accommodate those using the OGA 1.1 with ArkOS wanting to use the proper L2/R2.  By making a slight tweak to the .dtb file, L2 and R2 can be easily accomodated between the rgb10 and oga 1.1.  In the next update, I will provide a script in the options/advanced section that will allow people to switch this when they want, in the meantime you can make use of it by unzipping the attached file and place the .dtb file in the /boot partition.  You can either let it overwrite the existing file there or just rename the existing one just in case you may want to may want to revert for some reason.    The boot partition can be asked directly from a computer since it is fat32.
[rk3326-odroidgo2-linux-v11.zip](https://github.com/christianhaitian/arkos/files/7611043/rk3326-odroidgo2-linux-v11.zip)