christianhaitian/arkos #447: RG351MP not booting

D-Zlobin, 2021-11-05:
I have almost the same issue like this guy 

https://www.reddit.com/r/rg351mp/comments/qg07zh/arkos_not_booting_on_rg351mp/

With the only difference that I had a black screen instead of yellow.

I got latest available 09/30/2021 RG351MP arkos image, then wrote the image using Rufus to the brand new Sandisk Ultra microSDHC 16Gb Class 10 UHS-I which I bought in trusted store. I put it into device, switch on and got a blank black screen after short vibration buzzle. Then nothing happens for a long time.

I tried reboot several times, full device charge, put second SD card to device - without any luck.

I can provide any logs or try something else if you say so.

D-Zlobin, 2021-11-05:
Forgot to mention that stock firmware works fine, so I doubt that it is a hardware problem.

christianhaitian, 2021-11-06:
The 16GB Sandisk SD cards seem to give some folks issue with booting on the RG351MP.  Try a 32GB or larger Sandisk or Samsung instead.

D-Zlobin, 2021-11-06:
Wow, quick response! Thank you, I will try.

christianhaitian, 2021-11-06:
I also just uploaded an updated stable image.  Maybe that may help as well.

closed 2021-11-06 by christianhaitian

D-Zlobin, 2021-11-11:
It's me again. I bought 64 Gb SanDisk card and faced with the same behaviour as I posted before. Apparently it was not card size issue.

Now I'm using 351ELEC on the old 16 Gb Card and it works emmm ok. It boots well and launches some games I mean.

christianhaitian, 2021-11-12:
Made sure that only the main sd card was in the unit during boot?

christianhaitian, 2021-11-12:
Also, have you tried burning the image using USB IT? 
https://www.alexpage.de/usb-image-tool/download/

D-Zlobin, 2021-11-12:
Yep, I checked it again just now, the only card in device was my freshly burned ArkOs image, in slot TF1/INT.

>Also, have you tried burning the image using USB IT?

I didn't. I will try it tomorrow evening, thanks for your help.

D-Zlobin, 2021-11-12:
> Also, have you tried burning the image using USB IT?
> https://www.alexpage.de/usb-image-tool/download/

Unfortunatly the same result, boot logo then black blank screen forever.

christianhaitian, 2021-11-12:
Unfortunately, I have no more recommendations for a resolve at this time.

D-Zlobin, 2021-11-13:
Yeah, the device is new, and it's bringing new issues to developers. Thank you for your support!