christianhaitian/arkos #449: Online services not working - can't update

ransisco, 2021-11-07:
My RG351v do recognize wifi and can connect, it can upload retroarch cores, but everything online services (updating, scrapping, etc) that you'd use on the menu won't work.

christianhaitian, 2021-11-07:
For OTA updates, see this https://github.com/christianhaitian/arkos/issues/448.  Not sure what's causing your scraping issue though.  OTA wouldn't impact that.

ransisco, 2021-11-07:
I already tried the fix and didn't work.

christianhaitian, 2021-11-07:
Then something is up with your connection maybe.  Are you able to go to screenscraper.ft from another device connected to the same wifi signal?

ransisco, 2021-11-07:
I edited the comment saying that it fixed it. It even fixed scrapping

thx a lot pal ;)

christianhaitian, 2021-11-07:
no problem.

closed 2021-11-07 by christianhaitian