christianhaitian/arkos #450: Already unlocked achievements popping up every time I start a game

sammy2142, 2021-11-07:
Since the last update I've some issues with achievements.
Previously gotten achievements are displaying again everytime I load into game.
I've tried disabling retroachievemnets in retroarch etc and then re-entering my details which hasn't worked.

The achievements in question are showing up as already unlocked on the retroachievements.org website and in the retroarch menu. But as soon as a start the save game I get 3-4 popups of already unlocked achievements?
Ps. I haven't seen anyone else with this issue on the retroarch github. 
Regards

christianhaitian, 2021-11-07:
Best to report this on the retroarch github as that's not something I can control through ArkOS.

sammy2142, 2021-11-08:
Hi thanks for the reply - 
It looks like they rewrote the achievements code to make it multi threaded for version 1.9.13
I can confirm this bug happens in 1.9.13 on my desktop
It doesn't seem to affect 1.9.12 on windows
What commit was Retroarch based off in arkos as it seems to have a mix of 1.9.13 features but labeled as 1.9.12

Regards

christianhaitian, 2021-11-08:
That makes sense as I always build Retroarch on the latest updates available on the day that I build it.  You can see the commit information by launching retroarch and looking at the Git version.  For the last update, the commit is [96053e14a7](https://github.com/libretro/RetroArch/commit/96053e14a758c16156eb0fc9d42fba92a7004a0c)

christianhaitian, 2021-11-08:
Nice.  I see Retroarch is working on the issue

christianhaitian, 2021-11-08:
As soon as that's resolved, I'll get a new build out shortly afterwards

sammy2142, 2021-11-08:
Awesome thanks :)

christianhaitian, 2021-11-10:
New update is now available with retroarch 1.9.13 that includes the fix for the issue.

closed 2021-11-10 by christianhaitian

sammy2142, 2021-11-10:
Thanks - it works great :)