christianhaitian/arkos #453: List of compatible file formats/codecs for background music in the OS?

luhmont, 2021-11-11:
Don't currently have any background music that was converted from YouTube videos that will play in the background, regardless of the file type. This isn't surprising since the FAQ said as much about youtube to mp3 conversion songs, but it would be much easier to get around this if we had a list of compatible file codecs/formats, so that people can convert to the correct file for the music to play.

christianhaitian, 2021-11-11:
Per the FAQ, .mp3s file format works, just not ones from youtube.  Not sure why though as I haven't had issues with .mp3s working during my testing.  If you find a particular setting for mp3s that consistently works, let me know and I'll add that to the FAQ.

closed 2021-11-11 by christianhaitian

luhmont, 2021-11-15:
I used ffmpeg to inspect the file that was provided in the guide and then inspected a file that I got from a Youtube download. It seems that even though both say ".mp3" in Windows the files from YouTube are shown as AAC format when inspected. Looking at some of the other information provided it looks like this is because the audio was ripped from a video. I attached a few photos to show you what I mean, AUDIO1YT is from a youtube video, AUDIO2MP3 is the file provided in one of the guides I read, and AUDIO3SC is a file ripped from Soundcloud which works fine. I'm looking at finding a workaround now.

[image]
[image]
[image]

christianhaitian, 2021-11-15:
Interesting.  I wonder why the interest to get audio from youtube.  Is it very easy to do or considered the best source for audio nowadays?

luhmont, 2021-11-15:
Speaking for myself, it has the widest range of video game music for free coupled with the simplest search and download process. SoundCloud is decent but it doesn't have as much music.