christianhaitian/arkos #454: PPSSPP performance regression since being updated to 1.12.3 on RG351MP (update v2.0)

agorsar, 2021-11-14:
Hello,

I've been having some issues running some PSP games with the recent update to the RG351MP.
PPSSPP has been bumped to 1.12.3, and several issues have arisen.
The PPSSPP version in the menu is 'v1.12.3-185-gdf9be770d'.
I am using the 'Standalone' and 'Auto' options in the EmulationStation Emulation options for PSP, although 'Standalone-Stock' is worse than before as well.

While playing Jeanne d'Arc, the game is stuck at 8/30 fps, no matter what setting is used on PPSSPP. Setting frameskipping drops fps even more. This game ran much better before the v2.0 update. Disabling audio has no effect on performance. Even switching from dynarec to interpreted does nothing.
Even Half Minute Hero, which used to run full speed without frameskipping (60/60) sees drops to half or even lower frames per second. For instance, the white text scrolling on a black background after starting a campaign runs at 15/60 fps.
The display layout editor (which I've never used before) seems to be unusable right now.

PSP emulation on this device went from perfectly playable to literally unusable with the v2.0 update.

christianhaitian, 2021-11-14:
Hmm...yes I'm seeing issues with Half Minute Hero I never had before on my end as well with ppsspp 1.12.3 even with the later commits.  standalone-stock should not be an issue as it hasn't been updated and neither the assets it depends on.  Try the following.

Here's a backup of the previous version of PPSSPP (1.11).  See if this resolves the issue for you.

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1.

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed.  Reboot the device and test PPSSPP performance

If performance is still not back to where it was, you can try deleting the ppsspp cache files created by going to to your psp game folder, ppsspp\PSP\SYSTEM\CACHE folder and delete all the .cache files there.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/7534613/arkosbackup.tar.gz)

agorsar, 2021-11-14:
Did steps 1 and 2 and restores PPSSPP to 1.11. Made no difference whatsoever, although the emulator version is now on 1.11.
Tried deleting the cache, and the problems persisted.
The display layout editor is still unusable too.

Maybe something to do with '- Added missing backup config folder for ppsspp'? I remember I had to change something in /opt/ppsspp/ or /op/ppssppgo/ to make the settings save properly when changed, like button layout.

christianhaitian, 2021-11-14:
Nope.  The only time that comes into affect is if you delete your ppsspp sub folder within the roms/psp folder.  It will then restore the default ppsspp folder and contents when launching a psp game from the initial image.  Did you restore a previous backup of your settings from another installation of ArkOS on a different device?

christianhaitian, 2021-11-14:
As you stated that you were changing settings in ppsspp, perhaps you should try to delete that ppsspp subfolder in your roms/psp folder so it will go back to the default from the image.

agorsar, 2021-11-14:
That did the trick, I now have Jeanne d'Arc running at 100% (1x resolution) or 96% (2x res), with Half Minute Hero at a stable 100%.
I'd like to try reupdating PPSSPP to 1.12.3 and delete the folder to check if the performance problems stemmed from it. Do you have a backup archive for that?

christianhaitian, 2021-11-14:
Good idea.  Use the attached following the same instructions as before.  I'd be interested to see if that improves anything for you.  If the performance regresses again, just restore using the previously attached backup file.  Either way, please do respond with your results.  Thanks,
[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/7534764/arkosbackup.tar.gz)

agorsar, 2021-11-14:
Updated to 1.12.3 and the performance was retained. Problem solved.
Please take my boot and loading logos as a thanks!

[logos.zip](https://github.com/christianhaitian/arkos/files/7534783/logos.zip)

closed 2021-11-14 by agorsar

christianhaitian, 2021-11-14:
ooo clean and simple.  Thanks