christianhaitian/arkos #456: Solarus control mapping reversed

fakemaria, 2021-11-15:
### Describe the Issue (If applicable)
When playing solarus,analog stick controls are inverted (eg up is down)
And connecting a keyboard solarus menus aren’t showing up

### How can the issue be reproduced? (If applicable)

open a solarus game and play

### What device are you using?

Rg351v

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

latest update from system update

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2021-11-15:
I will address this in the next update of ArkOS.  This will involve going back to the last stable version of Solarus and not the current 1.70 development version..  For now, you'll have to fix your controls per game from their controls menu when possible.

fakemaria, 2021-11-15:
> I will address this in the next update of ArkOS. This will involve going back to the last stable version of Solarus and not the current 1.70 development version.. For now, you'll have to fix your controls per game from their controls menu when possible.

the problem is that I can’t enter this service menu either…
Can I have this bin? I can replace by myself

christianhaitian, 2021-11-15:
Which game are you trying to play?

fakemaria, 2021-11-15:
Ive rolled back to 1.7 bin (it was in the fw) and still same problem… can’t access to menu either with the keyboard plugged (unless its a combination of buttons im not aware of)

fakemaria, 2021-11-15:
> Which game are you trying to play?

Zelda book of mudora

fakemaria, 2021-11-15:
Ive tested more games and seems that some of them have problems with the implementation. So for me the manual rollback works,as some games allows you to map buttons,other don’t. Sorry to bother you,my first time with solarus ;)

Nice cfw btw!

christianhaitian, 2021-11-15:
Give me a moment and I will provide a solution for you to test.

christianhaitian, 2021-11-15:
Sorry, I don't have or use telegram.  Here's an update for Solarus that should fix the controls.  Just note that with this setup, the analog stick is still wonky but the gamepad should work fine.

1. Take the attached compressed file (arkosbackup.tar.gz) and put it in your roms/backup folder. Do not uncompress the contents! Just place the file as is in your roms/backup folder. If you don't have a backup folder in the roms folder, just create it.  If you're using a 2 SD card setup on your device, this must be done from SD1.

2. Then go into the Emulationstation Menu, scroll to Options, hit A, then go to Advanced, hit A, then scroll all the way down to Restore Settings, hit A.

3. Once completed.  Reboot the device and test Solarus

Let me know how it works out for you.

[arkosbackup.tar.gz](https://github.com/christianhaitian/arkos/files/7539911/arkosbackup.tar.gz)

fakemaria, 2021-11-15:
That worked! Closing the issue. Ive tested several systems and didn’t find any major issue (Im working alongside with Cisco :) )
Cheers

christianhaitian, 2021-11-15:
Thanks for confirming.