christianhaitian/arkos #459: Retrorun/32

fakemaria, 2021-11-17:
### Describe the Issue (If applicable)
A clear and concise description of what the bug is.
Retrorun and retrorun32 not loading because the /usr/local/bin/dremcast.sh is pointing to roms2 folder instead of roms

### How can the issue be reproduced? (If applicable)

Load an atomiswave game with retrorun32 and flycast rumble,sega logo appears instead of dreamcast circle

### What device are you using?

Rg351v

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen

latest from otg updates 

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2021-11-17:
Yea I see that.  You can simply fix the issue by doing a switch to sd2, then switch to main sd and it will take care of the issue until I can address it in a future update.

fakemaria, 2021-11-17:
> Yea I see that. You can simply fix the issue by doing a switch to sd2, then switch to main sd and it will take care of the issue until I can address it in a future update.

That drove me crazy xDDDD . Well it was the last system to test, I don´t know if there are more things?
But changing the sh is easier. Closing if you have it already in your backlog.

Cheers

closed 2021-11-17 by fakemaria