christianhaitian/arkos #462: Iconoclasts I'm curious about how to print other languages.

haveanair1, 2021-11-21:
You cannot change the language because the language setting does not appear. Do you know how?

JohnnyonFlame, 2021-11-21:
On 1.15 the language settings appear on first boot.

christianhaitian, 2021-11-22:
Hi, Have you confirmed that you're using the 1.15 version as JohnnyonFlame has stated?

haveanair1, 2021-11-22:
Hello. This is version 1.15. It is printed in English without a selection screen from the start. I'm looking for a way to initialize it.

christianhaitian, 2021-11-23:
So I'm running version 1.14 and on first launch, it gave me the option to pick a language (English, Deutsch, Francais, Espanol, and an Asian language).  I found that once you've picked a language, it saves your choice in the ports/iconoclasts/gamedata/lang.cfg file.  In order to be able to pick a language at boot again, you must delete that lang.cfg file.  You can do this directly on your unit using 351Files or just plug your sd card into your pc and manually remove it there.

closed 2021-11-23 by christianhaitian

soyer87, 2024-09-29:
Hi, I'm trying to play changing the language and I have downloaded some linux versions of Steam (1.14 and 1.15) but no one print language selector :S