christianhaitian/arkos #463: Does ArkOS undirectly supports N64 Hi-Res Texture Packs ?

prog-amateur2, 2021-11-22:
Hello, I searched the Wiki and its FAQ and could not find the answer to this question: 

> Does ArkOS _indirectly_ supports N64 Hi-Res Texture Packs ?

I would like to know if it was possible to edit a config file and give the path to the hi-res textures for the mupen64plus Rice and Glide64mk2 emulators. 

Thanks a lot for your help.

christianhaitian, 2021-11-22:
I think so by editing the mupen64plus.cfg file in the home/ark/.config/mupen64plus folder but i'm fairly sure they will negatively impact performance.

closed 2021-11-22 by christianhaitian

prog-amateur2, 2021-11-23:
Thank you ! Sometimes, the packs are light in term of size, so I hope some of them could be managed well. Anyway thank you very much for your kind reply, and good job.