christianhaitian/arkos #465: Atari 5200 Retroarch configurations

agorsar, 2021-11-26:
Hello again,

On my RG351MP, Retroarch configs have problems on the Atari 5200.
Once the game is loaded, retroarch loads with a specific configuration file located in home/ark/.config/retroarch/config/Atari800/, named "retroarch_5200.cfg", and not the system-wide configuration. This configuration can be changed, but cannot be saved (saving fails), and if a new config is saved, when an Atari 5200 game is loaded, that configuration is ignored and the old config is loaded instead.
Trying to load "retroarch.cfg" just crashes retroarch.
What's even stranger is that this configuration comes preconfigured for 3:2 screens. Maybe it's some sort of leftover from the RG351P/M?

christianhaitian, 2021-11-27:
The Atari800 emulator based systems do come preconfigured as they won't load games automatically unless you know what you're doing with them.  You can save changes for them as content overrides

closed 2021-11-27 by christianhaitian

agorsar, 2021-11-28:
Understood.
It's still very annoying to not be able to change and save settings such as hotkeys or retroarch UI.

christianhaitian, 2021-11-28:
Have you tried this using the overrides?

agorsar, 2021-11-29:
Yes I have. Retroarch UI settings (like changing to xmb) and hotkeys are not overriden.

christianhaitian, 2021-11-29:
Hmmm...ok then if you know how to configure atari800 emulators yourself, you can delete (or rename) the following configurations and set them up yourself:

/home/ark/.config/retroarch/config/Atari800/Atari800.cfg
/home/ark/.config/retroarch/config/Atari800/Atari5200.cfg
/home/ark/.config/retroarch/config/Atari800/AtariXEGS.cfg

Then you'll be able to save your configs how you want.

agorsar, 2021-11-29:
Yes I do. Thanks, I'll try that.