christianhaitian/arkos #468: RG351MP wifi not finding 5GHz access points

sksonic, 2021-12-07:
### Describe the Issue (If applicable)
I am on the latest update version of arkos for the RG351MP. I have a TP-Link AC600 USB Wi-Fi Dongle that supports both 5Ghz and 2.4Ghz wifi access points. When listing the wifi access points I only see the 2.4Ghz ones listed. The dongle itself works as I have tried running it with 351elec. The 2.4Ghz endpoints work well, but I am streaming games using moonlight so the 5Ghz endpoints are preferrable.

### How can the issue be reproduced? (If applicable)
Insert wifi adapter (usb c to usb 3) + wifi dongle. Go to wifi tool and list networks.

### What device are you using?
RG351MP

### What version of ArkOS are you on?
ArkOS 2.0 (11092021)

christianhaitian, 2021-12-07:
If it's this particular adapter (https://www.amazon.com/dp/B07PB1X4CN?psc=1&ref=ppx_yo2_dt_b_product_details), I have it and use it regularly with my RG351MP on the latest version of ArkOS 2.0 and it finds the 5Ghz networks just fine on my end.  If you have multiple SSIDs in your area, make sure to scroll through the list in case it was just further down.  If that doesn't work, try via the retroarch wifi settings and see if they show up there.

sksonic, 2021-12-07:
This is the same dongle yes, but purchased in the UK (it seems there might be differences in US/EU versions but I cannot confirm this). It is not a scrolling issue and also the endpoints don't show up in Retroarch. Comparing some dmesg logs in the emuelec versus arkos, the only difference I can see is the driver interface being loaded. When scanning the networks, I can see the wifi channels are pretty restricted on arkos.

Attached screenshots showing the driver and scan differences.

This is on the same hardware, at the same spot and has been consistent when repeated.
[image]
[image]
[image]
[image]

christianhaitian, 2021-12-07:
Same Product ID, SerialNumber, and Driver loaded on my end.
[image]
2.4Ghz and 5Ghz Channels show up for me.
[image]

christianhaitian, 2021-12-07:
What is your idProduct?  It should be the same I believe

sksonic, 2021-12-07:
Yes seems to be the same as yours:
[image]
.

sksonic, 2021-12-08:
This is the version of the module loaded on my version of ArkOS -> I originally had imaged the the test release of the RG351MP ArkOS and I subsequently ran the "update" tool on it to get to the latest version (in theory).
[image]

christianhaitian, 2021-12-08:
The update tool will get you to the latest version.
Same driver version on my end.
[image]

sksonic, 2021-12-08:
Seems to be related to a known issue with the driver version: https://github.com/aircrack-ng/rtl8812au/issues/850 

From reading this thread it seems the new release 5.6.4.2 did not help but another branch did https://github.com/morrownr/8821au-20210708 which might explain why it worked for me on 351elec. 

I'm guessing this can be closed as an issue with the driver - not sure how many people are affected by this and it might potentially a regional issue related to wifi channels. I might try figure out a way to load the alternative version of the wifi drivers if that is possible.

Side note: Thanks for the quick response and release of the pretty stable so far MP version of ArkOS. Really enjoying it.

christianhaitian, 2021-12-08:
No problem.  According to your pics, you seem to be using EmuElec not 351Elec.  Unless you just provided pics from stock.

closed 2021-12-08 by christianhaitian

sksonic, 2021-12-08:
Yes you are correct, the working version was stock EmuElec which I just happened to have on hand.