christianhaitian/arkos #470: Exult port lacks an on-screen keyboard, can't start new game

zomgugoff, 2021-12-10:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
The Exult Ultima VII port does not have an accessible on-screen keyboard. (I'm not sure if there actually is one). Also, There is no way to start a new game as the mouse pointer is hidden on the character creation screen. Originally, the mouse pointer would come back after pressing enter. On this port, enter is not mapped to a button, so there is no way to start a new game.

### How can the issue be reproduced? (If applicable)
Install as listed on the wiki, start the port, select either game, select 'START NEW GAME'. You can enter the keys that are mapped to the buttons, and highlight 'JOURNEY ONWARD', but none of the buttons work as a left-click on the mouse or enter.

### What device are you using?
RG351MP

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
2.0 (11092021)

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2021-12-10:
Response from the porter (nl255):
"Yes, that is a known limitation and unfortunately one that I don't think is fixable yet.  A note should be added to the wiki about needing to copy an existing saved game over though I do plan to test whether a real keyboard/mouse will work for that and will see about adding starting save data as well."

christianhaitian, 2021-12-11:
This information has now been added to the wiki entry for Exult as well.

closed 2021-12-11 by christianhaitian

romadu, 2021-12-13:
There is an update for PortMaster currently being tested that would resolve this limitation