christianhaitian/arkos #471: Fix? Restrictive Sound Card Selection/Design

justinnamilee, 2021-12-17:
So I have a USB Headset that I'd like to use, and through ssh / terminal I can make the required changes to ".asoundrc" to change from the default card to my USB headset (after disabling the various things that change and revert the .asoundrcbak / ford files temporarily).  This causes certain scripts to break like the handler for global hotkeys for volume, and the "current_volume" script, etc.

I'm on the RG351V.  I did much poking around but I didn't really understand the interaction of the oga_events.py and the async sound handler / how that was actually adjusting volume and such so I just left well enough alone.  Nor was I able to modify the original .asoundrc to use my new card and have that work, I'm much more familiar with pulseaudio as the audio server.

Like I said I can work around all of this via ssh then shut off remote services and play, but would be nice if there was a simple way to select the card and maintain hardware buttons, etc.  Maybe this is as simple as building a "correct" `.asoundrc`?  I've attached the OG vs the one that gets my headset running (card 1).

Dump of `aplay -l`:
```
(100% @ 3.4W) [amber->ark:~]$ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: rockchiprk817co [rockchip,rk817-codec], device 0: ff070000.i2s-rk817-hifi rk817-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Headset [USBC Headset], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```

[og_asoundrc.txt](https://github.com/christianhaitian/arkos/files/7731617/og_asoundrc.txt)
[new_asoundrc.txt](https://github.com/christianhaitian/arkos/files/7731618/new_asoundrc.txt)

christianhaitian, 2021-12-17:
Sounds like something interferes with ogage. https://github.com/christianhaitian/ogage/tree/rg351v

justinnamilee, 2021-12-17:
Yeah, 100% it is, same problem as the `current_volume` script, it's just hard-coded to use the `Playback` channel.

Almost need a script to select the Sound Card, perhaps added to the `/opt/system` folder (not sure how that populates to the RetroPie menu on the device itself, but yeah) then a second, wrapper script for the actual volume up / down amixer calls in ogage that reads which device to use on the card.

I'll whip something up and modify ogage to use it.   Any chance you can fill me in on how to add something to the "RetroPie" menu? Or do I just drop something into that system folder in `/opt` (I will try this, of course)?

Once I've got something I'll post it back here.

christianhaitian, 2021-12-17:
Yup, just add to `/opt/system` folder.  That's where Emulationstation looks for those types of scripts currently.

justinnamilee, 2021-12-17:
OK just writing the new wrapper to call in `emulationstation.sh` (to replace the `cp .asoundrcbak .asoundrc` part), but here's the card selection script `/opt/system/JL-SoundCard.sh`, and the helper generation script `/usr/local/bin/texmex` that parses `/proc/asound/pcm` for playback devices.

[SoundCardSelector.zip](https://github.com/christianhaitian/arkos/files/7737746/SoundCardSelector.zip) - Contains the two scripts.

Long story short it stores the chosen sound card in `/home/ark/.acard` for parsing by the wrapper script, what I'm doing now. :3

justinnamilee, 2021-12-21:
After much messing around with ogage -- where it turned out I just needed to compile on/for arm...(duh lmao) -- I have all the pieces in place.  Ended up dropping my /roms partition and expanding the root partition so I had enough space to install rust-minimal on the Anbernic, but we're here! I would say installing rust on device actually ended up taking the longest out of everything, haha.

```
(71% @ -1.6W) [amber->ark:~/source/ogage/target/release]$ sudo ./ogage
Channel selected: Playback
Added /dev/input/event3
Added /dev/input/event2
Added /dev/input/event0
Added /dev/input/event1
```

Now I'm not really sure how best to submit the changes, I could do a pull request on the ogage repo rg351v branch, or drop some files in here.  For now I'll drop the files in here and if you want them I guess you can use them.

First part is a selection script that runs from the Retropie menu, it handles picking the card, then picking the channel of the card and storing the results in a couple temp files.  Second is a small perl script that handles replacing the `.asoundrc` file.  Third is a few modifications to ogage's `main.rs` for reading the channel file.  Then the `current_volume` script just also needed to reflect the same change to read channel source from the file. And the final change is just to call the Perl script from `emulationstation.sh`.

The only caveat, I would say is that you have to reset the audio device to `card0` using the script before you can use the on board headphone port, but to me that's not really a big issue as I'm making a choice to change the audio card, so it makes sense I'd have to change it back to use that port.

Anyhoo, like I said above, I've attached the files below with a README that explains where they go and what they do.  My rust is...rusty at best, but it's a pretty minimal change so it should be fine.

Thanks for the pointers, feel free to close this, too.  Cheers!

[rg351v_changes.zip](https://github.com/christianhaitian/arkos/files/7758299/rg351v_changes.zip)

christianhaitian, 2021-12-21:
Thanks for sharing.  Since this seems like a limited use case, I'll leave this here for now.  I've seen many requests for being able to use bluetooth headphones.

closed 2021-12-21 by christianhaitian