christianhaitian/arkos #473: Feature request : kernel update plz?

ArcheyChen, 2021-12-17:
I found arkos 2.0 is a good os for my rg351mp, but the kernel is still a rather old version 4.4.
I first though this is the only version that rg351mp can run(it looks like odroid only released this version?)
but there is a other os call RetroLX(https://retrolx.org/), it use mainline linux 5.15.

I feel arkos has a way better experince then RetroLX, so can it possible import newer kernel to arkos?

I want to upgrade kernel to 4.9 or newer, mainly because I wanto apply a kernel patch call "Muqss"(http://ck.kolivas.org/patches/) ,it can make lowend device run much more smoothly.

but Muqss only support 4.9 -- 5.12 kernel.

and i tried to apply bfs(a scheduler before Muqss), to 4.4 kernel in wiki, but it fail to compile
(the main reason is that , bfs will need to disable the old scheduler, but somehow it still try to compile the old scheduler)
(and in order to compile bfs, kernel config must disable THREAD_INFO_IN_TASK option)

----

in short, i just want to add Muqss patch to kernel, and do a few twists to kernel config,such as:
- adjust the Timer frequency to 1000Hz
- enable Voluntary Kernel Preemption
- enable O3 optimization ( can be done on kernel 5.10)

----

also I wonder which version of kernel is rg351mp use? I can't find it on wiki.
and some kernel(such as 351p's) has a diffrent github  link in [click here] and [following the instructions here], which link should i follow?

christianhaitian, 2021-12-17:
As soon as the new kernel is stable and provides nice improvements for emulation, I will look to move ArkOS to it.  For the kernel used for the MP, see here: https://github.com/christianhaitian/linux/tree/rg351

closed 2021-12-17 by christianhaitian

ArcheyChen, 2021-12-17:
last one question,  is kernel bind to the chip(CPU), or to the board?
I only use linux on pc before, and i dont know how it works on arm devices.
 are the board related things written in kernel, or in the packages install on linux.

In x86_64, to install / replace a kernel, basically I only have to put kernel images in /boot  and place modules in it's directory.

but in arm, the file names looks quite diffrent , what files I have to focus on if I want to move the RetroLX's kernel to ArkOS?

christianhaitian, 2021-12-17:
Based on the other ticket you submitted, looks like you figured this out already.    The kernel is built against the SOC.  The SOC is based on more than just the chip, but also the controls and the video, power management, etc.