christianhaitian/arkos #476: PUAE Retroarch core virtual keyboard not accessible with control device type set to 'Joystick' or 'Keyboard'

zomgugoff, 2021-12-20:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
When using the PUAE Retroarch core, the virtual keyboard is not accessible with the control device type set to 'Joystick' or 'Keyboard'. It's much easier to bind keyboard controls with the control device type 'Keyboard' rather than in the core options (and cleaner for re-using the keymaps between similar games like pinball). This limit does not apply to UAE4ARM. I'd rather not use that core as it's quite limited compared to PUAE and the standalone emulator.

### How can the issue be reproduced? (If applicable)
-Load the PAUE core and start emulation, press select, the virtual keyboard will be displayed/removed.
-Enter Retroarch's menu, change the device type in controls to 'Keyboard' or 'Joystick', resume emulation.
-Pressing select will now not display/remove the virtual keyboard.

### What device are you using?
RG531MP

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
2.0 (11092021)

### Anything else you'd like to include that may help to review this issue or feature request?

christianhaitian, 2021-12-20:
Looks like there were some improvements to the puae core in relation to the virtual keyboard.  I'll build a new core and upload it to the repo to see if that resolves the issue.  I will report here when it's ready for you to do a core update in retroarch for you to test.

christianhaitian, 2021-12-20:
The core has been updated in the repo.  Please update the core through retroarch and report if it resolves the issue or not.

zomgugoff, 2021-12-20:
Sorry to say, it didn't change that behavior. Also, the default speed seems to be slower than before.

christianhaitian, 2021-12-20:
Looks like the keyboard is only selectable with retropad.  https://docs.libretro.com/library/puae/#virtual-keyboard

zomgugoff, 2021-12-20:
That's not what that page says.

"The PUAE core has a virtual keyboard that can be accessed by default through RetroPad Select."

It says "by default". Also, it gives controls for the keyboard input device. That seems odd, but I guess there must be a use case. Anyway, I just tried it with the Windows build of Retroarch and the virtual keyboard works on any input type (even 'none').

christianhaitian, 2021-12-20:
Hmmm...I think there's a small bug in the core as when I set the control to keyboard and set select to function as F11, virtual keyboard works.  No problem there, that's just a configuration needed.  However, when I then change the joystick to analog, virtual keyboard works.  Then I go back to the retroarch menu and set the control to just joystick, virtual keyboard now works.  That's a bug.

zomgugoff, 2021-12-22:
Any chance you could revert the core on the repo? The current one is too slow to be usable.

christianhaitian, 2021-12-23:
Done

christianhaitian, 2021-12-24:
Looks like the slowness issue was reported to the puae git here https://github.com/libretro/libretro-uae/issues/486
The update that caused the slowness has been reverted.  I will rebuild the core and update the core again since the older version that is there is still much newer than the core I went back to.  Your review and feedback on it’s performance will be helpful. I will post here when the update is done.

sonninnos, 2021-12-24:
"Joystick" is supposed to be a basic 2 button joystick, so it does not have any "Select" button to press. "Keyboard" does not either, therefore the toggle has to be routed via the keyboard mapping. "Keyboard" type also does not allow mixing joystick ports and keyboard inputs, which is why the core options exist.

christianhaitian, 2021-12-24:
Thank you for that clarification @sonninnos.  However, the user reports that the virtual keyboard works fine on any input on retroarch for windows using this same core.  How’s that possible.

sonninnos, 2021-12-24:
I don't quite understand. Sure it can be controlled regardless of the input device type when using actual physical keyboard, because it also listens to cursor keys and enter, not only RetroPad, but can't be toggled unless the device type has the toggle button mapped.

The joystick device type is pretty much obsolete now anyways, since the core can not anymore free RetroPad mapped keyboard keys as actual keyboard keys, since the blocking of overlapping mappings is now done by RetroArch.

So for example if you have cursor keys and control as a joystick, before one could disable those keys as joystick keys by putting the device type to None, and then the same keys would go through as keyboard events to the core. Now it does not work, because RetroArch currently acts based on if the keys are mapped or not, to prevent hitting multiple events at the same time. Joystick was there to have less buttons stolen from keyboard usage than RetroPad.

Same applies to the "Keyboard Pass-through" core option, which won't do anything anymore.

And indeed the keyboard device type is needed only for pinball games, since they are completely controlled with keyboard.

christianhaitian, 2021-12-24:
Thanks again for clarification.

zomgugoff, 2021-12-24:
The "Keyboard" device is not strictly needed, per se, as you can bind keys to controller inputs via core options, but that has limitations. There is no easy way to copy controls between cores in that manner as per-game core options cannot be loaded manually.

Currently, the only way to re-use core config controls is to copy a configured .cfg file in the 'configs' directory to another filename matching another game. Otherwise, you have to re-bind everything for each game, one at a time. The core config also cannot just be changed and re-saved either. The menu removes the save option when a per-game config is created (which is something that shouldn't be happening. You can re-save other defaults for things like shaders and remaps...). Instead, you have to remove the config, which reverts all of the settings, and you end up manually re-binding everything anyway.

Yes, this is almost entirely aimed at pinball games. It would be a lot simpler if I could just open the virtual keyboard and press a F-key to select a table or enter my initials on the high score table. But instead, I have to go into Retroarch's menus, change the input type, drop back to the game, open the virtual keyboard and do whatever, close it, go back into Retroarch's menus and change the type back. Or, I have to manually set up each game's controls, and re-bind everything when I want to change something (which also means re-binding everything on each game when trying to keep the controls as similar as possible).

I realize this is not specific to your build, @christianhaitian, these are problems in Retroarch for anyone that doesn't have a real keyboard connected to their device. I just thought it should be explained why I thought it was a bug and why it came up in the first place.

sonninnos, 2021-12-24:
No need for such trouble when the toggle is mapped to a keyboard key via core options, and that keyboard key is in the remap, for example in the same Select button.

F11 used to be doing that by default, but then I realized F11 is also the default key for RA mouse grab, and it was no longer needed after doing a separate key for it, so that when using a controller, that F11 mapping on the keyboard does not have to exist, since the controller button won't press F11, but does the same function what F11 would do if it was mapped.

opt:
`puae_mapper_vkbd = "RETROK_F11"`

rmp:
`input_player1_key_select = "292"` (292  = RETROK_F11)

christianhaitian, 2021-12-25:
@zomgugoff I've updated the core to the new reverted version.  Someone in the performance issue's ticket stated that the performance is better now on another Arm device.  Please update via retroarch and let me know if performance is at least the same as before.  Thanks

zomgugoff, 2021-12-25:
@christianhaitian It seems to be performing like it was on the old version. Thank you.
@sonninnos It seems that F11 is (now?) bound to the Virtual Keyboard in the core options by default, so only binding F11 to Select in the Input settings is necessary. Thank you for the workaround.

christianhaitian, 2021-12-25:
Thank you for confirming.  Closing this ticket as this issue can be considered resolved.  Merry Christmas!

closed 2021-12-25 by christianhaitian