christianhaitian/arkos #481: RG351MP No boot arkOS

luisalbes, 2021-12-24:
I have reviewed several messages and a couple of forums and the same thing happens to me that few people. 

The console cannot start the system, the arkos logo remains or the screen turns black. 

The version used is ArkOS_RG351MP_v2.0_11052021. 

It uses a Sandisk 128Gb extrem, a Chinese 16Gb sd and a Qilive 16Gb. Obtaining the same result with all. 

It is recorded with both ImageUSB, Rufus and Win32DiskImage. Obtaining the same result. 

It makes me think that it has a different chip revision or some bug in the kernel and therefore in certain consoles the system does not start. 

Could we review it in some way?

christianhaitian, 2021-12-24:
Try USBit https://www.alexpage.de/usb-image-tool/download/
Sometimes 16GB SD cards don’t work for the MP for the OS card.  I’ve had good results with a 32GB Samsung or Sandisk card for the OS (TF1) slot.

luisalbes, 2021-12-24:
I just tested the USBit program with 2 x 16Gb (Samsung and Qilive) sd cards and it still does the same. 

I only have to try a 32Gb sd, otherwise I will settle for something else. 

Thanks anyway, have a nice day

christianhaitian, 2021-12-24:
Sorry I couldn't be much help.  This unit can be more finicky that others for some reason.

closed 2021-12-24 by christianhaitian