christianhaitian/arkos #483: Update ArkOS error 500

madralf, 2022-01-03:
Hi! 
I tried to update ArkOS on my RG351M and got an error 500. A screenshot (well, photo) is attached. Seems like the update location (http://gitcdn.link/...) ist down..? Can't reach it on my PC either.

Maybe there is another way to update without completely wiping the data on the device?

Thank you!

christianhaitian, 2022-01-03:
gitcdn seems like it’s back up now..  You can also follow the instructions in the pinned issue [here](https://github.com/christianhaitian/arkos/issues/448) to use another method as well but I no longer update the 351p/m so the final update from May is all that’s available for that unit.

closed 2022-01-03 by christianhaitian

madralf, 2022-01-03:
Thank you for your reply! I still can't reach the site (not on the device and neither on my PC). I will just try again and maybe try your suggested alternative method.

christianhaitian, 2022-01-03:
For continued updates on this unit for ArkOS, checkout Wummle's fork.
https://www.youtube.com/watch?v=Y9wxFIOhn4g

madralf, 2022-01-05:
Thank you, will do!