christianhaitian/arkos #484: black screen after loading ArkOS on powkiddy rgb 10

ui-kideri-ui, 2022-01-08:
Hello, I installed ArkOS on my console. I followed exactly the instructions on the wiki, but when I turn on the console, I see a splash screen and then a black screen. What should I do

https://user-images.githubusercontent.com/97360001/148656689-844e7466-910a-4fe7-b248-5a23f7cc1719.mp4

?

christianhaitian, 2022-01-08:
What program did you use to image the sd card?
What is your sd card's make?

ui-kideri-ui, 2022-01-09:
i used win32diskimager, my sd card is 32 gigabytes, before that it had
emuelec

christianhaitian, 2022-01-09:
Was this the original card that came with the unit?

ui-kideri-ui, 2022-01-09:
yes, it came with the console

ui-kideri-ui, 2022-01-09:
I installed emuelec, the problem doesn't bother me anymore

closed 2022-01-09 by ui-kideri-ui

christianhaitian, 2022-01-09:
For future reference, the original sd card should not be used as they are problematic as stated in the installation instructions. Whatever distro you use, please use a quality sd card like Sandisk or Samsung.