christianhaitian/arkos #486: Feature request: noatime /etc/fstab

dreamer1234, 2022-01-16:
Hello:

I'm writing this because is a 0 effort change and because it's a change that it should be done.

/etc/fstab, should have the noatime option. Here's it why:
https://opensource.com/article/20/6/linux-noatime
https://stackpointer.io/unix/linux-io-performance-tuning-noatime-nodiratime-relatime/388/ 

This is a direct change (just to add the noatime for the ext4 filesystem). 
exfat filesystems, seem to also support this parameter: https://www.systutorials.com/docs/linux/man/8-mount.exfat-fuse/ 

Modern SSDs are very fast and most distros relay on default values (meaning a relatime value). In the past it was an important thing to consider. We are booting from SD card and the change is worth.

Thank you so much

christianhaitian, 2022-01-16:
Hello,

Did you check /etc/fstab?  It's already included for the main sd card 😉

closed 2022-01-16 by christianhaitian

christianhaitian, 2022-01-17:
Correction, I didn't add `noatime` it to the default ext4 partition for the system.  Although Ubuntu 19 mounts partitions using relatime which is not much less performant than noatime, it will save on unneeded writes to the Linux ext4 partition.  I will add this in the next update.  Thanks,