christianhaitian/arkos #487: On-Screen Keyboard Missing Symbol in Wi-Fi Setup

skullfireswords, 2022-01-18:
I am using ArkOS 2.0 (11052021) on an Ambernic RG351MP, and I am attempting to connect to my Wi-Fi. My Wi-Fi password has the "&" symbol in it, and this is not on the on-screen keyboard. Plugging in my USB keyboard to type this Symbol doesn't work either. Can this be added in a future version?

Thank you for your time.

christianhaitian, 2022-01-18:
Unfortunately the sources for the keyboard used for the wifi script is unavailable.  What you can do is use retroarch to connect to your wifi as that supports special characters.

closed 2022-01-18 by christianhaitian

christianhaitian, 2022-12-09:
Stays connected fine on my end when I configure it through Retroarch.  You can also go in via a local terminal connection and configure your wifi manually.  Check the FAQ for your device for more information on this.