christianhaitian/arkos #489: Error updating RGB10

tylibra2, 2022-01-23:
It's giving an error while updating. I click on update and an error message appears, it says that you are not connected to the internet. But I have internet working. RGB10

christianhaitian, 2022-01-23:
Are you able to scrape your games or download cores through retroarch?

SuperBadger77, 2022-01-23:
I have the same issue trying to update a Gameforce Chi and RG351MP

christianhaitian, 2022-01-23:
I've updated all three devices on my end without issue.  Has this (https://github.com/christianhaitian/arkos/issues/448) been already completed to get to the December 2021 update?

christianhaitian, 2022-01-23:
I think I see the issue.  The main CDN provider is having intermittent connectivity issues, just keep trying to update and it should eventually switch to one of the backup update locations.

tylibra2, 2022-01-23:
I have v2.0 (23/12/2021) Updated perfectly here!

Now in v2.0 (21/01/2022) it is giving error. It says it is not connected to the internet.

Now by RetroArch I can update normally and run the games. But, the RetroArch update is a little slow compared to the last times I updated. And there was a moment that the update froze, I had to cancel and come back again.

christianhaitian, 2022-01-23:
Yea see my previous response to the issue.  Just keep trying.

tylibra2, 2022-01-23:
It worked!

christianhaitian, 2022-01-23:
Yep. :)  Just the internet being funny today.

closed 2022-01-23 by christianhaitian