christianhaitian/arkos #494: Feature request: Segacd 32x easy loading

dreamer1234, 2022-01-31:
Hello:

Very few games were released for this plattform: https://segaretro.org/Sega_Mega-CD_32X . 
Problem I see is that there's not an easy way to load the games. 
I don't know if it could be possible to add an specific entry for this plattform to have a way to load the games easier.

Thank you so much.

christianhaitian, 2022-01-31:
Great request.  I’ll see what we can do.  In the meantime, it looks like they should load with picodrive.  Have you tried to load them through retroarch with picodrive yet to confirm functionality?

dreamer1234, 2022-01-31:
Thanks Christian.
To be honest, I have no idea. Those games were a combination of a key disc + game. First disc was needed to load before the game disc. The first disc is a sega cd disc (something like a protection) and the second one the segacd 32x. So for running the real disc (2nd), you need to load the 1st before.
It should work with picodrive. Have a look here: https://forums.launchbox-app.com/topic/50754-sega-cd-32x/

christianhaitian, 2022-01-31:
Update on this, I was able to play 2 Sega CD 32x games (Corpse Killer and Night Trap) fine from the existing segacd rom folder.  I just had to select the retroarch emulator then picodrive core in Emulationstation for Sega CD and they loaded fine.  I also made sure to have the proper bios files in the bios folder per the wiki for Sega CD.

christianhaitian, 2022-01-31:
So I don't see the need to create another system for this

dreamer1234, 2022-01-31:
I have tried with Farenheit. The emulation seems to work but if you insert the second disc with the game, it asks for the first one. So, perhaps is just this game that has that protection. 

Thanks you so much Christian.

christianhaitian, 2022-01-31:
Maybe a m3u file may help

closed 2022-02-01 by christianhaitian

dreamer1234, 2022-02-01:
I have found a workaround. 
Taken from here:  https://retropie.org.uk/forum/topic/27314/sega-cd-discs-m3u-failure

    Load disc 1 of Fahrenheit (key disc, .cue file) and wait until asks for disc 2 (game disc) (Picodrive does not support CHDs)
    While still on the Sega CD BIOS, open the RetroArch menu (select+X for the rg351mp)
    Scroll down to Disc Control
    Select Load New Disc
    Find Disc 2's CUE file in the file browser and select it.
    The game will load.

Now I see the audio has noise but that also means that the game works!
Thank you so much Christian.