christianhaitian/arkos #496: Bug: Bad sound Amstrad CPC crocods

dreamer1234, 2022-02-02:
Hello:
I'm having sound issues with amstrad cpc roms. Sound is broken for me and I'm asking because I don't know if this is a bug or it is something that happens with my installation.

I have updated Arkos to the latest update using the update option on the 351mpconsole.  I have also tried 351elec just to compare the sound and it is ok there. I have noticed they use the cap32 emulator (instead crocods) but I don't seem to have the option to choose it in arkos.

Can you have a look?

Thank you so much Christian

christianhaitian, 2022-02-02:
cap32 is available for Amstrad CPC.  Just change the default emulator for Amstrad CPC to Retroarch and the core to cap32.  You should also update the installed emulators for retroarch from within retroarch as well if you have't done this yet.

oaspiazu, 2022-02-02:
Sorry Christian. It is cap32 which is wrong and the one that has bad sound. Crocods does not seem to load the roms. I have updated crocoDS using retroarch's frontend (cap32 was already the last version) and I have still the same issues.

Thanks

dreamer1234, 2022-02-04:
Hello Christian. I have reinstalled everything using an older version and Amstrad works there. After upgrading retroarch cores and updating to the last version, amstrad cpc emulation is broken. I have only done those steps. So it is related to the update.
Thanks

christianhaitian, 2022-02-04:
Ok.  I’ll try this on my end and see what I get.

christianhaitian, 2022-02-04:
For cap32, looks like there was a fix here https://github.com/libretro/libretro-cap32/commit/5c24bc98eed1b3bc6fcad00de8ed7e14366c7bf3
I’ll get it built and pushed to the repo asap

christianhaitian, 2022-02-04:
I built the updated cap32 core and pushed it to the repo.  I've confirmed that it resolved the bad sound issue for Arkanoid and Ghosts N Goblins.  Also for CrocoDS, that emulator doesn't currently support compressed files like .zip or .7z.  I was able to play both Arkanoid and Ghosts N Goblins as .dsk files without issue.

closed 2022-02-04 by christianhaitian

dreamer1234, 2022-02-04:
It works!!! Thank you so much for the fix. Fantastic!!