christianhaitian/arkos #497: Feature request: overclock

dreamer1234, 2022-02-02:
Hello:
There's a very interesting thread here (retroOZ) about overclocking the console: https://forum.odroid.com/viewtopic.php?p=326429#p326429 

I have also seen that 351elec has also an overclocking option in emulation station and a dedicated kernel for that.

Could you add this feature? 

Thank you so much Christian.

christianhaitian, 2022-02-02:
It's placebo and doesn't nothing positive especially for 351 devices.  I had added it in the past and actually caused stability issues and in many cases slowed down emulation.  This is why I reverted the overclock in later releases.

closed 2022-02-02 by christianhaitian

oaspiazu, 2022-02-02:
Thank you so much Christian. In the rgb10 max seems to work

christianhaitian, 2022-02-02:
Try without enabling it and see if you notice a difference

oaspiazu, 2022-02-02:
Yes, at least in the rgbmax it works. Speaking with some people that has the rgb10 max, it seems that there's difference specially in Dreamcast emulation. In Saturn, you can win some few frames too. It's a bit tricky since to make the system faster than 1400Mhz, you also need to increase the voltage (between 0,95v and 1,35v in the case of the rgbmax).

Someone was also looking at the gpu but there's no info.

Thank you so much Christian

christianhaitian, 2022-02-03:
My understanding is that it’s not much better in performance however, it is more stable on OGS platforms.  For 351 devices, it does nothing but cause issues.  For the other oga clones, there’s been no benefit so far.

dreamer1234, 2022-02-03:
Heat dissipation on these devices is not good at all, so I'm guessing the problems are there. 
This mod is for the rg350M but is really worth: https://www.youtube.com/watch?v=bKhH14xpoqQ 

As allways, thank you so much for the feedback

dreamer1234, 2022-02-05:
Looking at this: https://github.com/351ELEC/kernel_rg351/blob/main/arch/arm64/boot/dts/rockchip/rk3326-rg351mp-linux-oc.dts  (cpu0_opp_table), it seems that the overclock of 351ELEC works at 1.328MHz, increasing also the microvolt value to which I understand the maximum and to keep the system stable (1,35v).  351ELEC seems to have 2 different configurations, so you can choose the overclock in ES. The overclocked system seems to work fine at that frequency.

The GPU is also overclocked in that configuration (gpu_opp_table). From 400Mhz to 560Mhz.

I understand you also increased the microvolt value. 
Thank you so much Christian