christianhaitian/arkos #498: More ext4 optimizations

dreamer1234, 2022-02-04:
Hello Christian

Have a look at this link: https://blogofterje.wordpress.com/2012/01/14/optimizing-fs-on-sd-card/ 

There are some pretty cool things explained about how to optimize the filesystem. At the end those read/write speed optimizations are also cpu cycles.

There's something he does not explain there. About the block size, some cards can do bigger sizes (i.e. samsung evo https://www.reddit.com/r/retroid/comments/icsw32/i_spent_entirely_too_much_time_researching/?utm_medium=android_app&utm_source=share ).  

Finally, I know it's not ideal for you but I have erased my exfat partition and replaced it for a full sized ext4 so I can have the best performance here. 

I think at least removing the journal thing and looking at those raid paramters is pretty interesting. 

Thank you so much.

christianhaitian, 2022-02-04:
Thanks Dreamer.  Be aware that this article is over 10 years old so many of these changes are implemented bu default already.  Also these optimizations will not lead to better emulation performance on this chipset which is the overall goal.  No problem going full ext4 on your end as it will lead to slightly better read/write performance p, however, it’s not something I eill implement as default since it would require many to have to result to loading their devices via wifi or a Linux OS which takes away from the ease of the system.

closed 2022-02-04 by christianhaitian

dreamer1234, 2022-02-04:
Thanks Christian.
Just to let you know that I did the change for a full ext4 and I have also disabled journalling since I don't think we need it in this device. I find the system very very fast compared to before. I don't know if it's me but even the boot time and everything is more fluid.
About journaling: https://forum.xda-developers.com/t/tools-how-to-disable-and-enable-journaling-in-internal-ext4-partitions.1252028/ 

It is not possible to make a patch for this since the ext4 filesystem can not be mounted to disable it , but the newest releases can have journalling off.

The rest of the parameters are specific to each sd card, so they do not apply here either. But someone can do it in his personal device using that tool to find the best way to format the sd card.

Thank you so much Christian

christianhaitian, 2022-02-04:
I'll check this out.  Thank you

christianhaitian, 2022-02-04:
Changing to full ext4 would have a performance boost in itself.  The journaling I will investigate

oaspiazu, 2022-02-04:
I understand that the full ext4 means that windows people need to be used to things like filezilla and send files over the network. Or perhaps with samba but the wifi dongle that comes with the console is really bad. I'm also using an usb-c to ethernet adapter for all the updates, which makes a super fast connection. Is working incredibly well.

Strictly, more read /writes (or a bottleneck due to the exfat use in linux), can imply more CPU cicles or load average if there are longer thread waits. So, performance can be worse at the end. It seems to me important to get ride of that exfat filesystem. Of course, I understand your point.

Thank you so much for all the support.

christianhaitian, 2022-02-04:
So I checked and journaling is already off for the primary ext4 partition for ArkOS.  So nothing to tune for that.  

[image]

Getting rid of the exfat filesystem would negate one of the few ease of use features of ArkOS so I just simply won't do that.  What performance penalty there is for accommodating exfat is very minor in my opinion.  Especially since it doesn't impact the emulation performance positively or negatively.  Of course, if someone wants to rid the exfat partition totally, they can do so on their own installs fairly simply using tools like gparted on Linux machines or minitool partition wizard on Windows machines.

oaspiazu, 2022-02-04:
Nice, I see. Perhaps it was my partition. 
Thank you so much Christian.