christianhaitian/arkos #504: ArkOs not booting properly on 351mp

drewbz1089, 2022-03-09:
Hi there,

 I have a RG 351mp and everything seems to work smoothly only when I boot it up it shows me the ArkOs screen, loads one circle and then goes to the black screen with a white dash at the top and just stays and never connects to emulation station.  After multiple restarts/power on/power offs every once in a while it will randomly load emulation station and everything boots fine. Anyone have any issues similar? Brand new RG351mp, and brand new SanDisk 128gb. I've switched microSD ports, updated ArkOs and am connected to WiFi. I used a MacBook Pro with ApplePi-Baker to originally flash ArkOs to the microSD(no issues with that).

Thanks!

drewbz1089, 2022-03-09:
also made sure to download the 351mp specific firmware prior to flashing drive

christianhaitian, 2022-03-09:
I’ve only had that happen when the sd card is problematic.  It can happen with a new sd card as well.  It’s rare but possible.  When the OS does boot, have you made sure to update ArkOS to the latest version via the OTA update?

drewbz1089, 2022-03-09:
Hit it with an OTA update last night when I was trouble shooting. I'm playing it now and it's working fantastic, just would feel better knowing its going to boot each time I want to go back in. Appreciate the quick response!

christianhaitian, 2022-03-09:
Ok.  I suspect part of the problem may have been the exfat filesystem driver that was previously being used.  It was less performant and had some other bugs.  It was updated with last month’s update and may help.  I also notice games load faster with this newer driver.  I’ll keep this ticket open for now. Please respond if you continue to have this booting issue.

drewbz1089, 2022-03-09:
Thank you, I'm copying everything from the original 128gb SanDisk over to a duplicate one that had already been flashed with ArkOs the same way at the same time.

drewbz1089, 2022-03-09:
Going to see if it changes anything

drewbz1089, 2022-03-09:
got everything switched over and updated again reinstalled roms/bios and still am running into the issue. It seem's its not wanting to connect to emulation station more than it does. Sometimes it takes a boot or two for it to connect but I've also ran into where it can take 4/5 times until it realizes emulation station.  Otherwise everything looks/works flawless when it does connect to emulation station.

christianhaitian, 2022-03-09:
Is this the case with both cards or just the one?

drewbz1089, 2022-03-09:
Both

christianhaitian, 2022-03-09:
Hmmm....Try another distro like 351Elec and see if the issue happens there as well.

drewbz1089, 2022-03-09:
Tried 351 Elec but I got a "Filesystem" error after flashing and booting up, it also wouldn't let me transfer ROMs onto the flashed 351elec micro sd. I've gone back and reinstalled ArkOs and re transferred everything back over because at least I can get it to work with a little elbow grease.

christianhaitian, 2022-03-10:
That’s not good.  You either have a flakey unit or bad sd cards.  Where did you get these cards from?

drewbz1089, 2022-03-11:
Still the same, starting to think it might be the device which isnt good because I cant send it back now that I’ve modified the SD cards

drewbz1089, 2022-03-11:
SD Cards were brand new SanDisk 128gb one from target one from amazon

christianhaitian, 2022-03-11:
hmmm.  Check with the seller on this.  Even if the sd cards have been modified or wiped, that may not impact the replacement process.

drewbz1089, 2022-03-11:
You think it would be worth it to just return it for a replacement via amazon?

agorsar, 2022-03-12:
Hi.
I also have this very same problem on my rg351mp.
This happens with two SD cards I have (Samsung 64GB and Sandisk 64GB, both A1 category).
Sometimes when turned on, it boots very quickly to Emulation station. Other times it can fail to boot, only showing a line ("_") on the console output. Sometimes, this line is orange, if the initial boot screen that shows ArkOS and dots below disappears while the last dot is orange. One or two resets later and it finally boots properly.
Sometimes the font size of that boot screen also changes (characters get bigger) while the dots are turning white and orange.

christianhaitian, 2022-03-12:
> Hi.
> I also have this very same problem on my rg351mp.
> This happens with two SD cards I have (Samsung 64GB and Sandisk 64GB, both A1 category).
> Sometimes when turned on, it boots very quickly to [...]

What version of ArkOS are you on?

agorsar, 2022-03-12:
I'm running the latest version.
This has happened with all previous versions too.

christianhaitian, 2022-03-12:
If version 02242022, and you're still running into that issue, you may want to try 351elec and see if it happens there.  If so, you either have a bad sd card or your unit is very sensitive to sd cards.  If you don't have the issue on 351elec, it may just be some sort of incompatibility with ArkOS.  I'm working on making a change to the kernel to see if it may help with that sensitivity issue.

christianhaitian, 2022-03-12:
Hi, can you try this file and see if it resolves your issue?  You just need to unzip it and copy the `rk3326-rg351mp-linux.dtb` file to the boot partition of the ArkOS card overwriting the existing file there.  It disables highspeed reading for sd cards which the MP may have problems with and in my few minutes of testing, it seems to be slightly faster in loading games now but it could just be placebo on my end.  Hoping it at least leads to more consistent booting for you.  Let me know your results either way.  Thank you,

[rk3326-rg351mp-linux.zip](https://github.com/christianhaitian/arkos/files/8238619/rk3326-rg351mp-linux.zip)

drewbz1089, 2022-03-13:
New 351mp received from Amazon this afternoon. Went out and bought a brand new 32gb microSD for the firmware and another 126gb for roms.  I've got everything flashed and working smoothly this time, everything works correctly/properly as far as the ArkOs firmware is concerned but for some reason when I went into advanced settings and switched it to run the 2nd microSD for roms it won't display the Systems in the lobby. Any idea how I can correct this? If I can get this issue solved I should be smooth sailing finally

christianhaitian, 2022-03-13:
Did you reboot yet?

drewbz1089, 2022-03-13:
seems to work fine when I put the bigger microSD 126gb in the first port by itself running ArkOs with the roms folder on the same microSD(but with no micro sd card in the second port). When I have the smaller MicroSD(32gb) in the first port with just ArkOs flashed to it(go into advanced settings and change it to run the second bigger microSD for roms) it doesn't display the systems/games. Just to see what would happen I also tried running the bigger in the left side and the smaller in the right but it still doesn't want to bring up the systems in the lobby

drewbz1089, 2022-03-13:
reboot tested and I get a file error that says "looks like your systems configuration file has not been set up or is invalid, you'll need to do this by hand, unfortunately" and then it says to visit emustation.org for more information

christianhaitian, 2022-03-13:
First off, the second sd card can not have multiple partitions on it.  It must be formatted to one partition which a new card normally is.  If you have an OS flashed to both sd cards, that will not work correctly. Only flash one sd card with arkos for the system, then the second sd card must be formatted to either exfat or ntfs.  It can have roms preloaded on it on the one partition but must not be flashed with any other os.

drewbz1089, 2022-03-13:
ok cool I'll check it and see if it works

agorsar, 2022-03-13:
Tried the .dtb file provided, and cycled power ten times.
Before, it would need a reset (or two) 50% of the time the console was powered off. Now, it needed a reset the first time it booted, but the remaining 9 booted correctly.
During the boot the font changing might still occur (happened three times out of ten, including the first failed boot).
It appears that when it also boots _faster_ now than before (when it successfully booted, of course).
One last tidbit that might be interesting, regarding the font size change. When powering down, the font size and type is differente than when booting, and has always been this way, with the boot color being an orange, and the powering down being a kind of yellow. The powering down font and color is much nicer, I must say.

agorsar, 2022-03-13:
Interestingly, I reverted to the "old" .dtb, and the hanging during boot has stopped. Couldn't recreate it. Could it have been a corrupted .dtb file?
I'm sticking to the normal .dtb file for now, and report if it happens again in a day or two, I guess.

drewbz1089, 2022-03-13:
After erasing the firmware on the larger 126gb MicroSD and reformatting to exfat and carrying all the roms over, I power up and go into advanced/switch it to 2nd SD for roms, reboot and I’m still getting the “ "looks like your systems configuration file has not been set up or is invalid, you'll need to do this by hand, unfortunately" and then it says to visit emustation.org for more information” error

christianhaitian, 2022-03-13:
You're issue is very odd.  Have you made sure to complete online system updates before enabling the 2nd sd card?  It shouldn't matter but I'm not sure why that issue is happening for you.

christianhaitian, 2022-03-13:
Also, you make sure not to restore any old backups?

drewbz1089, 2022-03-13:
Brand new cards bought yesterday. I’m going to go get one more new one today on my way home to work in case flashing/erasing may have corrupted the one I got yesterday just to see

drewbz1089, 2022-03-13:
Still have the two original SD’s the newer rg351mp came with untouched in case none of this pans out

drewbz1089, 2022-03-13:
Finally got it going perfect on both SD’s! Thank you so much for your patience with me and helping me troubleshoot!

agorsar, 2022-03-15:
Just as a follow up on my part.
Put the old *.dtb file, ran into the same boot problems again. Haven't had a single problem with the file you sent though.
And the game booting seems as fast as ever, on my machine.
Thanks Christian.

closed 2022-03-16 by christianhaitian