christianhaitian/arkos #505: PortMaster new update crash

RobertBalogh, 2022-03-24:
There is a new PortMaster update available on Anbernic RG351P (Thank you)
But sadly if you download the update, your PortMaster will fail to start. 
Guru meditation #68B329DA9893E3409

Shadow53, 2022-03-24:
I have also encountered this.

FWIW, a fix is to download PortMaster from its repository and manually extract the files to `/roms/tools` (or `/roms2/tools`), matching the existing layout -- this fixed it for me. I don't recall if I had to make any modifications to the script. I don't think so, but I might have forgotten.

christianhaitian, 2022-03-24:
Guru meditation?  There's no such in ArkOS.  Did this happen on 351Elec?  Also, I haven't encountered issues with updating on any officially ArkOS supported devices.

Shadow53, 2022-03-25:
Further context on my part:

- Latest update of arkOS was installed (I think)
- Opened PortMaster from the tools menu
- PortMaster prompted to update itself. I selected yes.
- PortMaster updated itself.
- PortMaster would no longer start
- Manually update PortMaster from the zip on its repo
- PortMaster now runs just fine

christianhaitian, 2022-03-25:
I can't replicate this issue.  The only thing I can think of is that between portmaster version 4.6 and 4.7, the portmaster.sh file is moved one folder above the portmaster folder so that 2 steps are not needed in order to get into the portmaster menu.  If you were updating from portmaster 4.6, it could be that just a simple restart of emulationstation would have resolved the error.

RobertBalogh, 2022-03-28:
Actually I could sort the issues. 
Updated manually with copy paste the new version of the PortMaster. 

I believe there's some error happened when I tried to update through the PortMaster. 
Tried a few times, but same issues came up. 
I use 351ELEC (updated) on my device. 

Thank you for your hard work christianhaitian. 

I hope we can see more game ports available in PortMaster in the future.

closed 2022-03-28 by RobertBalogh

Daddelidu, 2022-04-28:
Hi, my Problem is that i can´t confirm the Update. It comes on (Anbernic RG351M) ELEC351 and ArkOS the Message" There´s an update for PortMaster (6.4). Would you like to download it?" But i can´t confirm it. No matter which button i press. It seems that there is an Issue with the controls or buttons?
[image]

RobertBalogh, 2022-04-28:
You should update it manually I'm afraid. 

Copy and paste is the way.

Daddelidu, 2022-04-28:
Stupid question, but where do i get latest version? I downloaded it from here: https://amberelec.org/PortMaster

RobertBalogh, 2022-04-28:
I was searching for the latest version on Google. 
But maybe you can find it on Github. 

Hope it helps.

Daddelidu, 2022-04-30:
Unfortunately i still don´t get Portmaster running. Always Update Issue. Any Idea or Link to Version 6.4?

christianhaitian, 2022-04-30:
https://github.com/christianhaitian/PortMaster/raw/main/PortMaster.zip

christianhaitian, 2022-04-30:
Instructions are here https://github.com/christianhaitian/PortMaster

Daddelidu, 2022-04-30:
Thank you very much for your fast support!!!
But i don´t get it. I have already read the instructions and do it exactly the way in the instructions. I´ve deleted the old folder and unzip new PortMaster folder from your Link to my Ports folder. But it is still the same problem with the Update. I am not able to press yes and confirm update.

christianhaitian, 2022-04-30:
It shouldn’t be asking for an update if the contents were unzipped already as that zip file contains the latest released version already.

Daddelidu, 2022-04-30:
Now it works! Finally ;-)  I zipped the PortMaster.Zip in my Downloads Folder and not directly in the PortsFolder. That was the Problem.