christianhaitian/arkos #506: Hangs at black screen with blinking dash

nasiralamreeki, 2022-03-24:
Ive followed all the steps and after inserting the SD card it shows an ArkOS boot sequence then some sort of dialog mentioning easyroms with an ok bottom pops up for a few miliseconds I can barely read and then a black screen with a blinkinh white dash.

Ive checked the card there is a boot and easyrooms partition if I put rom files in and then reboot the OGA 1.1 it still stalls at the black screen with blinking white dash

christianhaitian, 2022-03-25:
Are you using a name brand sd card?  How did you image the sd card?  What image did you download and use?

nasiralamreeki, 2022-03-25:
Yes I'm using a Sandisk I've tried two others I've tried instructions  on Mac and Windows both have the same result.

I'm using the one from the wiki as it doesn't appear any of the new build images are available anywhere it's build v1.5 (12/26/2020)

christianhaitian, 2022-03-25:
all builds from the wiki are 2.0 except for the one for the rg351p which is a final version.  Are you sure you downloaded the one for the oga 1.1?  They're all available at the bottom of the wiki here: https://github.com/christianhaitian/arkos/wiki

nasiralamreeki, 2022-03-25:
Yes I'm certain I downloaded the OGA 1.1 build

christianhaitian, 2022-03-25:
Then you don't have a 1.5 build.  Please post the link you used.

nasiralamreeki, 2022-03-25:
https://drive.google.com/file/d/1yi2PeUfoCqiprhQFJkPpk6IsGtWdxRVs/view?usp=sharing

christianhaitian, 2022-03-25:
That is the current link.  Try to image the file onto the SD card using USB IT (https://www.alexpage.de/usb-image-tool/download/).

christianhaitian, 2022-03-26:
Hi @nasiralamreeki,  I've uploaded a new updated image to test and see if it will help with your case.  The link to is [https://drive.google.com/file/d/1B42Tx_AK0puCsIzxUQ9p2c5Sba8uA7Kx/view?usp=sharing](https://drive.google.com/file/d/1B42Tx_AK0puCsIzxUQ9p2c5Sba8uA7Kx/view?usp=sharing).
Please try this image and see if this works better.  

Thank you,

christianhaitian, 2022-04-02:
Hi,

Any update on this?

Thanks,

christianhaitian, 2022-04-16:
Closing due to no further updates on this from the op.

closed 2022-04-16 by christianhaitian