christianhaitian/arkos #511: Request for BBC Micro emulator on RK3266 devices

SuperBadger77, 2022-04-29:
Hi,

I have previously asked if it is possible to get a BBC Micro emulator running on RK3266 devices. Now I've had it up and running on Recalbox on my RGB10 Max 2: https://wiki.recalbox.com/en/emulators/computers/bbc-micro. So should be something that runs on the RG351 series. Is this something you could look at?

christianhaitian, 2022-04-29:
I would review the possibility of adding it but the source recalbox uses (https://gitlab.com/recalbox/recalbox/-/blob/master/package/beebem/beebem.mk) is not available publicly.

SuperBadger77, 2022-04-30:
I've emailed the owner to see if he can assist.
http://www.mkw.me.uk/beebem/ Mike Wyatt beebem.support@googlemail.com

SuperBadger77, 2022-04-30:
It does state on his website: http://www.mkw.me.uk/beebem/
The latest version of BeebEm has been released by Chris Needham and the latest sources are now maintained in his repository on [GitHub](https://github.com/stardot/beebem-windows). The source code is released under the terms of the [GNU General Public License](http://www.gnu.org/licenses/gpl-2.0.html).

christianhaitian, 2022-04-30:
The version recalbox is using is https://gitlab.com/bkg2k/beebem-linux.  That’s the version that would need to be used since it’s sdl2 based and available for Linux.

SuperBadger77, 2022-04-30:
Ah ok - raised the query on the Recalbox discord: https://discord.com/channels/438014472049917953/438796001059274784/969922038082990090

christianhaitian, 2022-04-30:
Seems like an interesting system so I'm curious to see how to make it work.

SuperBadger77, 2022-04-30:
It is an interesting system! Some of the first ARM RISC processors come from this computer \ company. In the UK they were practically in every school, so remembered fondly by people of a certain age ;-). Massive game library with some classics such as Elite. I do hope we can get this running :-) I'll keep you posted if I get a response from discord.

SuperBadger77, 2022-05-01:
Response from the author:

Hi Rik,

Good to hear from another retro and BBC Micro enthusiast!  I'm all for widening the audience for retro projects and BeebEm in particular.

BeebEm has already been used in a number of bundled emulation systems.  The issue of licensing does always raise some questions.  Back in 2009 I got the agreement from all of the BeebEm authors to switch the licensing of the Windows build to GPL2.

The Linux build of BeebEm though, which a lot of these third party systems use, pre-dates this GPL change, and no-one (to my knowledge) has yet updated the sources to bring them in-line with the latest Windows version.  So, licensing for the Linux sources remains as the original author (Dave Gilbert) specified in the COPYING file (attached).  Having said that, I'm sure Dave would have no problem with BeebEm being included in an ArkOS build, as long as it was freely available.

More of an issue is licensing for the BBC Micro OS and BASIC ROM images.  Although some effort has been made to establish who owns the copyright for these ROMs, it has never been fully resolved.  There remains a risk that a commercial entity may at some point object to the use of the ROMs (although no-one has ever contacted me regarding use of them).

The Linux port of BeebEm is available here:

  http://beebem-unix.bbcmicro.com/

There may well be some further updates available for the Linux port scattered across the Internet - I've never looked in any detail!

I've not seen recallbox before.  It looks like they picked up the Linux port and, even though they have kept their repo private, I doubt they have made any significant modifications to it.

Hope all of this helps.  If you do port BeebEm to a new system, I would be happy to add a link to it from my BeebEm site.

Kind regards,
Mike

christianhaitian, 2022-05-01:
Recalbox made some changes to the sources to make it sdl2 compatible.  When I tried this, it wasn’t as simple as changing every instance of sdl-config to sdl2-config in the configure file within the sources.  Without knowing what was done or someone assisting me with providing  guidance on this who’s more familiar with sdl coding, I’m stuck at the moment.  Perhaps the person who merged the build to recalbox, bkg2k, will eventually open it for others to benefit but, who knows.

SuperBadger77, 2022-05-02:
I've raised a post on the Recalbox forum as well: https://forum.recalbox.com/topic/26825/help-required-porting-the-bbc-micro-emulator

SuperBadger77, 2022-06-03:
Hi, I’ve tried several times to get someone from Recalbox to share the code but had no response. Tried on discord and their forum. No dice so far. Will keep trying. 

I did however notice that the BBC Micro and Acorn Electron is now available on several Anbernic devices on Batocera v34. They appear to use MAME MESS to achieve this. Hoping something similar could be possible for ArkOS? Was also chatting to Slaminger on Discord and he said it should be doable.

let me know what you think?

SuperBadger77, 2022-06-03:
https://wiki.batocera.org/systems:bbc

christianhaitian, 2022-06-04:
I was hoping to get BeeBem as configuring Mame for things like this a headache.

SuperBadger77, 2022-06-04:
Ok, I’ll keep pushing.

SuperBadger77, 2022-06-08:
I’m still trying: https://discord.com/channels/438014472049917953/439141063425392640/984233889058914324

SuperBadger77, 2022-06-14:
I've left one last comment: https://forum.recalbox.com/topic/26825/help-required-porting-the-bbc-micro-emulator/6 assuming I don't hear anything back within the next few days, I'll close this request off :-(

SuperBadger77, 2022-06-22:
@christianhaitian when I get time I'll look at trying to fix the repo Konsumer created: https://github.com/konsumer/beebem to see if I can get that working, then I'll try also building ArkOS myself and seeing if I get anywhere with that. Just need the time!
Not sure if I've raised this before but could you look at this repo: https://github.com/stardot/b-em It's a unix version of B-em - also a BBC emulator. Quite advanced and well supported and on github it says its under GNU GENERAL PUBLIC LICENSE v2.0. Would this one work on ArkOS?

christianhaitian, 2022-06-27:
I had tried to build b-em and ran into an issue I don't recall that involved the Allegro.

SuperBadger77, 2022-06-27:
bkg2k finally got in touch, and he wasn't happy with me harassing him :-( which I think was unfair. He didn't respond to any of my queries until now, and someone else said previously he didn't want to share it so not sure what I was supposed to think. All he had to do was say he was working on it and will make it public when ready:
https://forum.recalbox.com/topic/26825/help-required-porting-the-bbc-micro-emulator

I haven't responded to him yet but the upshot is that he was planning to make the repo public but was just trying to find time to fix the sound issues. So the version we were after may become public imminently if not already.

christianhaitian, 2022-06-27:
Ok.  Thank you for the update.

SuperBadger77, 2022-12-06:
@christianhaitian what's the best way to get in touch with you to discuss progress? Discord? Could do with some help / queries. I've had some help from @konsumer trying a few new approaches. He created this repository: https://github.com/konsumer/beebem/ to get the unix build of beebem working in SDL2.0. I managed to get this working on a 64bit Raspberry Pi OS build. It runs but it didn't seem to have any sound and the input was very weird - maybe a timing issue.

After some more help from @konsumer, I took https://github.com/tom-seddon/b2 (which is actively updated) and I've almost got this building on a 32bit Raspberry Pi OS build - very close.

If I can get either of these up and running on Pi OS (Buster Debian) under SDL 2.0 - would they then be candidates for inclusion in ArkOS? I assume there is more work to be done after I get them running un SDL 2.0 in Unix? Any help or guiders appreciated.

christianhaitian, 2022-12-06:
Hi, I'm most available on the Retro Game Discord.  You can find the link on the same page as the links to download the various images.  If you want to try running the SDL 2.0 build on ArkOS, I recommend getting another sd card and loading ArkOS on it, then enable Developer mode https://github.com/christianhaitian/arkos/wiki/Building-packages-and-modules-on-your-device.  Then build and test it on the device directly in ArkOS.

konsumer, 2022-12-07:
My SDL2 build was an untested shot-in-the-dark experiment with sdl-compat (SDLL1 to SDL2 compat-layer) but I can't really vouch for it, and didn't test at all. The SDL2 port is probly a better path than my terrible hack.

That said, we also found [b2](https://github.com/tom-seddon/b2) which looks maybe a bit nicer, is written for SDL2, and the only serious-looking issue I found was some [timing stuff](https://github.com/tom-seddon/b2/issues/207) which looks fairly fixable.

I don't think I have enough bandwidth to help too much right now, but I wish you well, and later, when I have more time, I'd love to help.

christianhaitian, 2023-03-19:
Closing this issue for now since we're at a standstill until a viable solution is available.

closed 2023-03-19 by christianhaitian

christianhaitian, 2023-03-27:
ArkOS uses gptokeyb <https://github.com/christianhaitian/gptokeyb/tree/main>.  Mess is available in the core repo used by ArkOS.  You can try it there and see how it works.

SuperBadger77, 2023-03-27:
Actually just mounted the INT card and found es_systems.cfg

christianhaitian, 2023-04-03:
no, unless you setup a log for the emulator running process itself.  I would go in via a ssh terminal session, stop ES `sudo systemctl stop emulationstation` and try running the BBC emulator manually via the ssh terminal session and see what happenes.

SuperBadger77, 2023-04-03:
I created this entry in es_systems.cfg
```
	<system>
		<name>bbc</name>
		<fullname>BBC Micro</fullname>
		<path>/roms2/bbc/</path>
		<extension>.zip .ZIP .bin .BIN .7z .7Z .ssd .SSD .bbc .dsd .adf .ads .adm .adl .fsd .wav .tap .bin</extension>
		<command>sudo perfmax %EMULATOR% %CORE%; nice -n -19 /usr/local/bin/retroarch -L /home/ark/.config/retroarch/cores/mess_libretro.so %ROM%; sudo perfnorm</command>
		<platform>bbc</platform>
		<theme>bbc</theme>
	</system>
```
Which didn't work - so I followed the instructions, logged on, stopped ES. Then tried this manually:
`sudo perfmax ; nice -n -19 /usr/local/bin/retroarch -L /home/ark/.config/retroarch/cores/mess_libretro.so /roms/bbc/Airwolf.ssd; sudo perfnorm` and getting the following output (not sure where the error is).
[output error.txt](https://github.com/christianhaitian/arkos/files/11141722/output.error.txt)

christianhaitian, 2023-04-03:
Try the following in terminal and see what you get:
`/usr/local/bin/retroarch -v -L /home/ark/.config/retroarch/cores/mess_libretro.so /roms/bbc/Airwolf.ssd`

SuperBadger77, 2023-04-03:
[output error 2.txt](https://github.com/christianhaitian/arkos/files/11141969/output.error.2.txt)
Getting further.

christianhaitian, 2023-04-03:
mess doesn't know what bbc is.

christianhaitian, 2023-04-03:
This mess version is 3 months old.  Was BBC support recently added to mess?

christianhaitian, 2023-04-03:
do you have the necessary bbcb.zip file in the bios folder?
https://wiki.batocera.org/systems:bbc

SuperBadger77, 2023-04-03:
BBC should have been added years ago. I thought I’d added all the bios to the external card but this was running the game off the internal card (the manual execution). Ive been using the Batocera bios pack so it works. If I’m on a single card where does the bios go? I’ll double check everything and try again tomorrow. I might ask Slaminger for some pointers as I think he has it running on Retro Arena.

SuperBadger77, 2023-04-03:
I’ll do some more research into Advmess: https://forums.raspberrypi.com/viewtopic.php?t=23113 is this the same thing?

christianhaitian, 2023-04-03:
advmess is standalone and not a core.  Maybe that's what is needed?

SuperBadger77, 2023-04-03:
Is advmess available in ArkOS?

christianhaitian, 2023-04-03:
> Is advmess available in ArkOS?

Not currently.

SuperBadger77, 2023-04-04:
Looks like Slamiger was using Advmame: https://discord.com/channels/452235057982078987/845054469917245512/1084173327754608843 
Is it possible to get Advmess added to ArkOS? Or can it be installed on an image to test?

SuperBadger77, 2023-04-04:
Left a comment on the RA discord: https://discord.com/channels/452235057982078987/845054469917245512/1092705587478941758

SuperBadger77, 2023-04-04:
> sudo systemctl stop emulationstation

Double checked the bios - it is there. Same error still.