christianhaitian/arkos #513: Feature: able to run raylib games, and notes on building things

konsumer, 2022-05-23:
This is related to building things (#294)

I didn't end up getting raylib to work. Ray suggested `PLATFORM_DRM`, but I got this:

```
INFO: Initializing raylib 4.1-dev
INFO: Supported raylib modules:
INFO:     > rcore:..... loaded (mandatory)
INFO:     > rlgl:...... loaded (mandatory)
INFO:     > rshapes:... loaded (optional)
INFO:     > rtextures:. loaded (optional)
INFO:     > rtext:..... loaded (optional)
INFO:     > rmodels:... loaded (optional)
INFO:     > raudio:.... loaded (optional)
INFO: DISPLAY: No graphic card set, trying platform-gpu-card
INFO: DISPLAY: Failed to open platform-gpu-card, trying card1
INFO: DISPLAY: Failed to open graphic card1, trying card0
WARNING: DISPLAY: Failed to find a suitable DRM connector mode
FATAL: Failed to initialize Graphic Device
```
Do you know of any tricks to make framebuffer stuff work with things like this?

Here are some notes for future builds:

I am using a mac M1 (arm64), so I didn't need binfmt-support/qemu-user-static installed, but you can use those on amd64, if you need it.

```sh
docker run --platform="linux/arm64/v8" -it --rm -v $(pwd):/workdir -w /workdir ubuntu:eoan bash

cat << EOF > /etc/apt/sources.list
deb http://old-releases.ubuntu.com/ubuntu eoan main universe
deb http://old-releases.ubuntu.com/ubuntu eoan-updates main universe
deb http://old-releases.ubuntu.com/ubuntu eoan-security main universe
EOF

apt update
apt install -y gnupg

echo "deb http://deb.odroid.in/go2 bionic main" >> /etc/apt/sources.list
apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 5360FB9DAB19BAC9
apt update
apt upgrade -y

# build raylib

apt -y install git nano build-essential cmake libasound2-dev mesa-common-dev libx11-dev libxrandr-dev libxi-dev xorg-dev libgl1-mesa-dev libglu1-mesa-dev libgles2-mesa-dev libgbm-dev

git clone https://github.com/raysan5/raylib.git
cd raylib/src/
make PLATFORM=PLATFORM_DRM
cd ../examples
make PLATFORM=PLATFORM_DRM
```

christianhaitian, 2022-05-27:
Where did you try to execute this from?

konsumer, 2022-05-27:
Not sure I understand.

I built it in docker (so I have all the proper headers and stuff) and copied the compiled program, copied onto my device, and ran there, in ssh. Docker was faster for me to setup than spinning up a vmachine, etc, and it works fine for compiling other things.

The built program runs fine, but looks like it expects X, so it's probly some setting to make it use the framebuffer (with gles.) maybe it just doesn;t work at all that way. I will play around with it more.

closed 2022-06-08 by christianhaitian

konsumer, 2022-06-08:
Ended up getting it! I had to kill emulationstation, and make sure the game was 640x480