christianhaitian/arkos #514: NAOMI/ATOMISWAVE WON'T LOAD

Orion81, 2022-05-26:
Sorry for my English
I use rgb10 with ark os 2.0 with last update. Also retroarch and retroarch32 updated.
Naomi and Atomiswave bios  are located in the / bios / dc folder.
After load rom splash screen ark os, black screen, splash screen ark os an return to emulation station.
I've tried every core.

christianhaitian, 2022-05-27:
Hi, I tested SalaryMan on Atomiswave on the latest update using Retroarch32 and flycast32_rumble core and it started fine for me.  I tested Noukone Puzzle Takoron on Retroarch32 and flycast32_rumble core and it started fine for me as well.  Have you tried updated the retroarch cores in the retroarch menu to be sure you're on the latest version?  If so, you may have a bad bios.

Orion81, 2022-05-27:
Is there a specific romset or bios or rom format?
Retroarch is fully uploaded and the bios are in the folder roms/bios/dc

christianhaitian, 2022-06-02:
Any update on this issue?

christianhaitian, 2022-06-04:
Closing as I can't replicate the issue and there's been no further updates.  Verify your bios and roms files are of good quality is all I can recommend.

closed 2022-06-04 by christianhaitian

Orion81, 2022-06-08:
Wrong romset. Now it works