christianhaitian/arkos #515: RG351MP stuck after splash screen on first boot

eduardantonio, 2022-05-31:
### Describe the Issue (If applicable)
Hello, Christian, I hope you are doing fine!

I wanted to report an issue that I'm currently observing in a RG351MP device.
I have been trying to install ArkOS using the download items from [the wiki](https://github.com/christianhaitian/arkos/wiki) using both links Mega and GDrive (specifically for the RG351MP version), but so far I always get stuck after the splash screen.

I have followed the steps from the wiki:
1, I have used a SanDisk Ultra (from Amazon) and two different PYNY (from BestBuy) SD XC cards (all of them 64gb) but facing the same result.
2. I have prepared the SD cads using  USB Image Tool, Win32DiskImager  and even ETCHER (sorry, I read the warning in the wiki, but just in case :D)
3. I insert the SD card in the RG351MP (no other SD card is inserted in the TF2/EXT slot)
4. And then turn the device on, but always experiencing the same behavior:

https://user-images.githubusercontent.com/39350451/171078876-9f62bb5e-3f13-4c46-9fe8-355ee984db53.MOV

After splash screen is dimmed, the device gets stuck in the black screen that you can see from the video.
I was reading [this thread ](https://github.com/christianhaitian/arkos/issues/504) and I have tried with the `rk3326-rg351mp-linux.zip` file that you shared in there, but still seeing the same behavior. (I also read from the same thread that another user reported good results after going back to the previous `rk3326-rg351mp-linux` file it worked, but unfortunately on my side this was not the case) 

Is there any other version that I can try for the device (maybe an older version and then do an update?)
Or if there any way to debug the script files to see at what step the device is getting stuck?

I 'm sorry about the bother, I know this has been asked before, but just wanted to double check in case there has been any news about the RG351MP device.
Please let me know if you need something from my side and I will be able to provide it.

Thanks a lot for your time!

### What device are you using?
A RG351MP from Amazon.

### What version of ArkOS are you on?
Trying to install `ArkOS_RG351MP_v2.0_11052021` from [GDrive ](https://drive.google.com/file/d/1iq_38i_hrMCRnflQ_UuXjYtuEVOlp5Fp/view?usp=sharing)and [Mega](https://mega.nz/file/2Y5WHDJb#aQvXTd34WaUtEnZydOcqkdkFhDHR3Bx7j3wtpJXY8zQ).

christianhaitian, 2022-05-31:
I'm still working on resolving this issue for some rg351mp units.  I was able to acquire a unit that exhibits this issue so now I can replicate it which is helpful.

eduardantonio, 2022-06-01:
That sounds good, and thanks a bunch for the heads up! 
If you need anything from my side, please don't hesitate to let me know.
Have a nice day! :D

AriFarugie, 2022-06-20:
Hello all,
I have the same issue and I hope you will fix it soon :)
I am very thankfull christian for your effort :D

christianhaitian, 2022-06-20:
I need to get back to working on this unit.  I've been having fun with the rg503 lately.

reaperofpower, 2022-07-10:
Following this, Same issue :(

christianhaitian, 2022-07-10:
For those that can't boot the RG351MP using ArkOS, please take the rk3326-rg351mp-linux.dtb file that's in the attached zip file and copy it to the BOOT partition of your ArkOS sd card overwriting the existing one in there.  Then see if you're able to boot into ArkOS.  I was able to boot ArkOS on the unit I have using this updated dtb file.  It disables the ability for the RG351MP to even think it's capable of 1.5Ghz as it seemed to cause the unit to not be able to boot.

Please report whether it's successful for you or not.

Thank you,

[rk3326-rg351mp-linux.zip](https://github.com/christianhaitian/arkos/files/9079429/rk3326-rg351mp-linux.zip)

reaperofpower, 2022-07-10:
> For those that can't boot the RG351MP using ArkOS, please take the rk3326-rg351mp-linux.dtb file that's in the attached zip file and copy it to the BOOT partition of your ArkOS sd card overwriting [...]

That did it, it now booted and allowed me to complete the setup

christianhaitian, 2022-07-10:
Excellent.  Thank you for confirming.  I've updated the available image for the RG351MP on the wiki with this fix and the latest updates pre included.

closed 2022-07-10 by christianhaitian

eduardantonio, 2022-07-11:
Thanks a lot, Christian! I can also confirm that now this is working fine in my unit! Really appreciate the fix 😁👌

christianhaitian, 2023-01-05:
That check is a carry over from the Odroid Go Advance and Super by HardKernel and has nothing to do with these Ambernic units.  They just copied them and left them there apparently.