christianhaitian/arkos #516: RGB10 Pro Moonlight invert abxy

gricardo93, 2022-06-02:
Hello,

Is it possible to set Moonlight from Portsmaster to invert abxy?
I'm using a RGB10 Pro with latest ArkOS and it's kind of annoying because there's no option in Moonlight to invert abxy.

christianhaitian, 2022-06-02:
Do it from the portmaster settings menu

gricardo93, 2022-06-02:
Yeah, I did it, but it only inverted X and y. I had to change manually gamecontrollerdb.txt to invert a and b.

Thank you for your response.

Do you know if it is possible to allow local audio using this streaming option?

https://github.com/moonlight-stream/moonlight-embedded/wiki/Usage

christianhaitian, 2022-06-02:
It is but you'd have to edit the launch script manually to add that.  I forgot that I still need to fix the button invert issue for the PortMaster script.  I'll scheduled that to be done asap.

closed 2022-06-02 by christianhaitian