christianhaitian/arkos #518: Game closes and returns to menu

WillianBelmon, 2022-06-08:
For some reason my ArkOS closes the game if I just open the retroarch menu while playing. This issue happened after update my ArkOS on RGB10.

WillianBelmon, 2022-06-08:
> What game system and game were you playing when this occurred?

I tried almost all game system that supports retroarch cores, they all having the same issue. I've played castlevania sotn, super mario world, chrono trigger, sonic 2. These games works just fine but they all crash when I try to open the retroarch menu inside the game. It never happened on previous version

christianhaitian, 2022-06-09:
I can’t seem to replicate this issue.  Was this a sd card you setup with ArkOS yourself or was it a preloaded one?

christianhaitian, 2022-06-09:
It sounds like the exit hotkey needs to be changed.

christianhaitian, 2022-06-09:
The default to exit retroarch is minus key+start(2 times).

WillianBelmon, 2022-06-09:
> The default to exit retroarch is minus key+start(2 times).

Dude I found a fix for that, I just needed to reset the retroarch settings.

WillianBelmon, 2022-06-09:
Seems like the ozone menu is the problem after the update

christianhaitian, 2022-06-09:
I just tested again with the ozone menu set as the menu driver, still can't replicate the issue.

christianhaitian, 2022-06-09:
Is this issue resolved for you at this point then?  Can you use the Ozone menu without the content closing on you when you load the retroarch menu?

WillianBelmon, 2022-06-10:
> Is this issue resolved for you at this point then? Can you use the Ozone menu without the content closing on you when you load the retroarch menu?

No, I still cant use Ozone menu but rgui, xmb and glui are working fine to me