christianhaitian/arkos #519: Wifi Freezing issue

Manurocker95, 2022-06-11:
### Before submitting your issue or feature request.  DO NOT REQUEST ASSISTANTS FOR PRELOADED IMAGES.  I do not support or condone the use of preloaded ArkOS images.  Preloaded meaning images that come with copyrighted roms preincluded.  Look for support from the creator or supplier of those preloaded images, not from here!  If a ticket is submitted for assistance for such images or if there's an indiciation that a preloaded image is what the support request is for, the issue ticket will be closed with no further response.

1. ***This is not Anbernic Stock Emuelec or 351Elec.***  ArkOS is a different distribution from those systems and will have differing features.  The only somewhat similarity is that they both use EmulationStation as a frontend.  Even with that, the versions of EmulationStation that's used in between ArkOS and Anbernic Emuelec/351Elec is different with different capabilities.  If there are features you prefer with Anbernic Emuelec/351Elec, just use those distributions instead.  The choice is yours.

2. Have you reviewed the Frequently Asked Questions available at https://github.com/christianhaitian/arkos/wiki/Frequently-Asked-Questions---RG351P ?

3. If your issue is related to being able to run certain emulators or ports, have you reviewed the Emulators and Ports section available at https://github.com/christianhaitian/arkos/wiki/ArkOS-Emulators-and-Ports-information for needed bios files and supported rom extensions?

4. Have you reviewed the ***known issues*** section available at https://github.com/christianhaitian/arkos/wiki/Known-issues ?

### Describe the Issue (If applicable)
A clear and concise description of what the bug is.

Whenever I try to connect my device to wifi using the wifi dongle that was included with the rg351MP, the entire device freezes completely requiring a full restart, pretty similar to what happened before as you can see here: https://github.com/christianhaitian/arkos/issues/96

In my case, even on the main menu, once I connect the dongle, arkOS gets stuck completely.

### How can the issue be reproduced? (If applicable)
Flash and install ArkOS 2.0 in a new RG351MP and try to connect to wifi with the "default" dongle.

### What device are you using?
RG351MP

### What version of ArkOS are you on?
- You can find this during the initial boot process right before the Emulationstation loading screen
2.0

### Anything else you'd like to include that may help to review this issue or feature request?
It was working on my first try, I followed the steps to "keep ssh to always remain enabled" and then boom. Stopped working :/

Manurocker95, 2022-06-11:
Charging the battery to 100% doesn't fix the issue. I don't have any other dongle so haven't tried "other types" either

christianhaitian, 2022-06-11:
You don't state what sd card you're using so I'm assuming it's an authentic Samsung or Sandisk sd card unless you state differently.  It's possible something may have went wrong with the included dongle.  Does work on the stock os without issue?  You can also try a fresh image, then setup your wifi dongle again.  If everything works, make sure to do the OTA updates so you're on the latest updates.  Be sure to do all this with nothing else connected to the otg ports but your wifi dongle.  Not even one of those magnetic charging adapters if you use something like that.

tristanhampton, 2022-06-17:
Hey Christian! I seem to be running into the same issue that was referenced (96). I'm using a higher quality SanDisk SD card, and I have a dongle and USBC to USB adapter that I'm using to connect to wifi.

The dongle worked perfectly when I used 351elec on my 351p, but if I try to set it up now it freezes. Any thoughts as to why this would be?

I'm at 100% charge, I've reflashed the sd card a handful of times using different softwares on my mac (apple pi baker and balena, but most recent is apple pi baker).

christianhaitian, 2022-06-18:
Is this on the 351MP?

tristanhampton, 2022-06-18:
This is just the 351p model!

tristanhampton, 2022-06-18:
Sorry, I guess it's a slightly wrong channel but the issue is the same 😅

christianhaitian, 2022-06-18:
I'm not supporting the 351p so can't help on that one.

closed 2022-06-18 by christianhaitian