christianhaitian/arkos #521: EasyRPG is broken (RG351V)

MaxiMuM1441, 2022-06-24:
Hello there!

I recently updated to the latest version of ArkOS on my Anbernic RG351V. Before that I believe I was on version 11/09/2022. Since the update, I am not able to start any RPG Maker game. When trying to start a game, it opens the EasyRPG app and I need to select a game again. All games end with .easyrpg. When launching a game, it crashes and I'm back on Emulationstation. I already ran the Scan_for_new_games script and checked the gamelist.xml file and the directions seem to be correct. All games are located under /roms2/easyrpg. Of course, RPG_RT.ini and RPG_RT.ldb are not missing. It worked fine before the update.

christianhaitian, 2022-06-24:
Did you update the easyrpg core in retroarch?

christianhaitian, 2022-06-24:
If not, do that and see if you're able to launch your games then.

MaxiMuM1441, 2022-06-24:
Updating the core in RetroArch fixed it. Thanks for the help! I didn't know that I also have to update the cores in RetroArch. I thought new cores already come with the latest ArkOS update.

closed 2022-06-24 by MaxiMuM1441

christianhaitian, 2022-06-24:
Excellent.  No I don't update cores during updates just in case it causes issues for folks.  I let folks choose to do that on their own.  Enjoy!