christianhaitian/arkos #522: Feature Request: support wireless display

qianhd, 2022-06-25:
Hello, Christian, 
Hope you are doing well!

I wonder if there is any possible for ArkOS to support wireless display.
I know there is a lot of wireless display technology on all kinds of cellphone, like Airplay on iPhone, Miracast on Android phone.
Using these technologies, we can enjoy games of the big screen instead of  the 3.5-inch screen.（Maybe the display quality is not goot due to the resolution）

Thanks a lot for your hard working. Please reply me at your convenience

christianhaitian, 2022-06-25:
Hello, None of the devices arkos supports will be able to do this.  Best solution is hdmi out of the rg503.

closed 2022-06-25 by christianhaitian

qianhd, 2022-06-25:
Hello , 
Is it a technical obstacle or have no plan do this? rk3326 performance is too weak to support ?

christianhaitian, 2022-06-25:
Technical limitation of the chipset.

qianhd, 2022-06-25:
> Technical limitation of the chipset.

Thanks